wled/WLED #1749: Segments Wizard

data-toaster, 2021-02-15:
**Is your feature request related to a problem? Please describe.**
Sort of. I'd like a visual helper to identify segments, when creating them. In my case I have words, and I'm passing the LED strip through the letters. I'd like to isolate words in segments, for better lightning, but because of the design it's difficult to understand which one, for example, is LED 100.

**Describe the solution you'd like**
Ideally, in the segments panel, a button to light LEDs sequentially, with a counter telling you the number of the LED currently lit. This way I can note down the first and last LED for the segment I want to identify. 

**Describe alternatives you've considered**
Manual counting... not optimal at all.

**Additional context**
n/a

pa1okz, 2021-03-01:
Principally, all LEDs within the newly defined segment in question gets lit as soon as you hit the update (V) button. In that way it should be rather easy to figure whether the definition should contain more or less LED's, right?

data-toaster, 2021-03-01:
That is a way, yes. 
Another option would be as something even more interactive: you either know already the number of the LED you want to stop on, so you type it in, or you press a button like "NEXT" and the LEDs are sequentially lit. Once you reach the one you want to stop to, you can perhaps press another button and it'll be set, or you read the number and type it in manually

blazoncek, 2023-08-05:
As a help, you can use Percent effect.