wled/WLED #1750: WS2812B Question

Doyle4, 2021-02-15:
Hello, 
Really sorry in advance as its not a WLED question, but more of a WS2812B question and I cant find anything on it else where and hoping someone with knowledge could me out.

Id like to purchase individual WS2812B's, but I cant figure out what the little SMD is on WS2812B's strips, capacitor? is this essential to use or can it be missed out? Individual WS2812B's are so cheap, but pre soldered single ones are more expensive and I can get X3 the amount for same price single.

Thanks, and sorry again :)

sansillusion, 2021-02-15:
Yes it is a capacitor and it is needed for signal integrity. The LEDs can
consume a lot of power in fast spikes and that could become a problem.

Doyle4, 2021-02-15:
@sansillusion 
Thanks very much for answering.

closed 2021-02-15 by Doyle4