wled/WLED #1751: WS2811S 12V

yogi1325, 2021-02-16:
Any ideas how to run the WS2811S 12V strip, as the diode trick work only on 5v strips?

Thank you in advance

toto79, 2021-02-17:
My idea: Data-Line with 5V works on most 12V Strips, like ws2815. See [picture](https://github.com/toblum/McLighting/issues/351) for wiring.

dmxlaser, 2021-02-20:
As i found  on my own setup, do not use logic transistor level shifters with ws2811, they cause flickering some times.
If its needed use the 74hct125 or the 74ahct125.
Level shifters with PNP transistors or PNP FETS are a very bad choise.

koenslootmans, 2021-03-02:
Hello dmxlaser,

I am experiencing the problem that you reference too I belive, I have a 5m 12v led strip with in total 150 leds hooked up.
There is levelshifter on the dataline (DVAR-SPK11978 from sparkfun) , the strip is powered with a 5amp power supply (and the same supply is powering the wemos d1 mini that drives the led via a voltage reducer. all grounds are connected)

The strip works ok untill a certain point when I increase the brightness then the strips starts flickering. (depending on the choosen animation the brightness level at which flickering appears is different, it seems the more leds are lit at a given time the lower the brightness must be kept to not have the flickering effect.

My questions are the following:
- Is this indeed the behaviour when the level shifter is not ok ?
- the level shifters you are refering too are not on boards, how can I solder them ? (I am a newbie so this is probably a dumb question)
-

Thanks for any help much appreciated.
Cheers

ghpk, 2021-03-06:
> Hello dmxlaser,
>
> I am experiencing the problem that you reference too I belive, I have a 5m 12v led strip with in total 150 leds hooked up.
> There is levelshifter on the dataline (DVAR-SPK11978 from [...]

Looks like a clear case of Low Power to the LED's

Try powering the strip from both ends and it should be fine.

dmxlaser, 2021-03-08:
> Hello dmxlaser,
>
> I am experiencing the problem that you reference too I belive, I have a 5m 12v led strip with in total 150 leds hooked up.
> There is levelshifter on the dataline (DVAR-SPK11978 from [...]

Hi there.
The DVAR-SPK11978 from sparkfun was my first choise and also was the problem.
You can try to find a schematic with the sn74hct125n and solder it by yourself or 
try the TXS0102  breakout board if you don't 
want something like the DVAR-SPK11978 from sparkfun.

koenslootmans, 2021-03-08:
Hello, Thanks for the comment

I just managed to fix the problem yesterday, I soldered all unused ports on
the board to the ground and the problem is now gone,
The lights are now 100% stable

Cheers,
Koen

closed 2021-07-15 by stale[bot]