wled/WLED #1752: Thunder effect with API

toppe, 2021-02-16:
**Describe the bug**
There is a problem with my WLED setup. Currently i'm using Node-Red to switch it with the motion sensor. In the middle of the night when there is motion it turns on with the thunder effect instead of the "red" colour it supposed to be

**To Reproduce**
Non. When i'm tyring to reproduce it's switches as expected (just turn on)

**Expected behavior**
Only turn on with no effect. The POST/GET URL used to switch it on: http://192.168.1.137/win&A={{payload}}&FX=0&TT=1000. Where {{payload}} is a value between 0 and 255. Again: During 'day' time everything works fine!

**WLED version**
 - Board: ESP
 - Version 0.10.0
 - Format Binary

Thank you for your help!

Aircoookie, 2021-02-18:
Hi!
There is no thunder effect in WLED - do you mean "Lighting"? I believe the problem might be due to LED flickering rather than the incorrect effect being applied. I'm afraid I can't help much if the issue is not easily reproducible - next time it happens you can access the UI and see if it is on Solid effect as expected or on a wrong effect.

toppe, 2021-02-18:
Hi,

Sorry, yes, it's the "Lightning" effect.

I'm using Node-Red with a motion sensor with lux sensor. I've used a range node with a input range between 0 and 160 and a output of 1 and 255 (the output is the brightness). 

I've tried to reproduce the same situation as in the middle of the night it's shows that the lux value is not between 0 and 160 but between -1 and 160 with the result that the output is invalid. Is there any check if the parameter is negative?

Aircoookie, 2021-02-18:
Hi!
Please check what NodeRed actually outputs to WLED :) The -1 is technically not a legal input value for the brightness, but it is mapped to 255 in my testing. You could also try omitting the FX=0 value from the API request, there is definitely something weird going on here...

toppe, 2021-02-18:
The current solution is to change de range node from 0/160 to -1/160. I need to check the debug information from NodeRed when the -1 'bug' pop-up's. Will check it tonight and give you more information when i've got it!

toppe, 2021-02-18:
@Aircoookie Just as expected, it's possible to use a negative value in the `&A` parameter. The light turns on very high. Maybe it's possible to check if any value on the API is negative to round it up to 0?

closed 2021-06-28 by stale[bot]