wled/WLED #1753: ESP32 DevKit not connect to wifi

arigia62, 2021-02-16:
I have completed the flashing of two ESP32 Dev KITs using ESPHome-flasher and after the reboot while seeing the WLED AP network on my mobile I cannot connect.
After entering the password, the network disappears from the list and reappears after a few seconds.

The router is located about 1 meter from the board and the fixed red LED and the flashing blue LED are on on the card.

The problem is present whether the board is powered via USB or applying 5V to the VIN and GND pins (5V - 2A power supply)

For completeness, I report the ESPHome-flasher screen shot at the end of the flashing
[image]

Doyle4, 2021-02-17:
Unplug the module, I get this error sometimes but still works after.

arigia62, 2021-02-18:
Hi Doyle4, I followed your advice but the problem remains (I also tried with an RSP8266 -F12 Dev Kit). In the list of available wifi networks, WLED AP appears but as soon as I enter the pw (wled1234) it makes me the message "it is not possible to connect to this network" and the network disappears from the list

Doyle4, 2021-02-18:
Try using this flasher.
https://github.com/nodemcu/nodemcu-flasher

Baud set to 460800
Flash 40mhz
SPI DIO

Model-Man, 2021-02-18:
ESP home flasher does not format and partition a ESP32 correct and can cause issues. As advised try a different flasher then check your available memory. It should be around 1500Kb.

closed 2021-06-28 by stale[bot]