wled/WLED #1756: expansion of Per-segment individual LED control JSON

mogim0gi, 2021-02-17:
**Is your feature request related to a problem? Please describe.**
I want to make 8 leds in the start of the segment to turn red on a switch, the rest of the segment should continue whatever activity is going on.

**Describe the solution you'd like**
expand JSON with parameter that would allow this

**Describe alternatives you've considered**
I have tried :

curl -d '{"seg": [{"id": 0, "i": [[255, 0, 0], [255, 0, 0], [255, 0, 0], [255, 0, 0], [255, 0, 0], [255, 0, 0], [255, 0, 0], [255, 0, 0]]}]}' -H "Content-Type: application/json" -X POST http://192.168.1.179/json

**Additional context**
I guess this is still doable with current functionality, but it is quite complicated:

turn on :
1. GET current state of segments
2. POST a new 8 led segment + adjust other 2 segments length
3. POST above JSON targeting newly created segment

turn off:
4. Cleanup segments and apply GET from step 1

Thank you for your ideas for making WLED better!

Aircoookie, 2021-02-18:
Hi!
I agree this would be a useful addition. Currently the entire segment is disabled when using individual mode, this is because the LEDs set to red would immediately be overwritten by the effect function otherwise. We'd need an extra buffer to tell the effect which LEDs not to update.

Currently the easiest way to achieve it would be shortening segment 0 with `{"seg":{"id":0,"start":8}}` and setting up the segment 1 to red. Afterwards, delete segment 1 again and put segment 0 back to start at index 0 using `{"seg":[{"start":0},{"stop":0}]}`. This saves you having to GET and store the entire state and thus makes your application logic simpler :)

mogim0gi, 2021-02-18:
Thanks, ill try it out tomorrow. I imagine this low-level control of strip could open many new ways of wled application.

flok, 2021-08-30:
I'm also really interested in this feature. I currently try to display diffrent variable from my home automation on the end of the strip.  Did you have any success @mogim0gi  ?

blazoncek, 2023-08-05:
Solution provided above.
Closing.

closed 2023-08-05 by blazoncek