wled/WLED #1757: HUZZAH 32 /feather IO2

Yyov, 2021-02-17:
Hello, 

i am using a Feather Huzzah 32. 

The requested port is not brought out of the board. i read that it is possible to change these settings but i lack the knowledge to recompile the .bin fille. 

can i get some help with this? 

Kind Regards! Yov

srg74, 2021-02-18:
You can use Discord #bot to compile custom firmware for your board. If you not able to do yourself ask someone and you will receive help.

Yyov, 2021-02-19:
> 
> 
> You can use Discord #bot to compile custom firmware for your board. If you not able to do yourself ask someone and you will receive help.

Thx, as i said, i lack the knowledge to do this so help would be welcome :) as i can see in the other questions it seems like a easy job for someone who knows what he is doing!

closed 2021-06-28 by stale[bot]