wled/WLED #1758: ESP32 Devkitc_V4 does not start after flashing

rlfnb, 2021-02-17:
**Describe the bug**
I have an ESP32 Devkitc_V4 (ESP32-WROOM-32) and downloaded the bootloader and the WLED binary from the releases page as described.

**To Reproduce**
`
(esp32) :~/workspace/python/esp32$ esptool.py erase_flash
esptool.py v2.8
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b4:e6:2d:eb:1b:c9
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 11.7s
Hard resetting via RTS pin...
(esp32) :~/workspace/python/esp32$ esptool.py write_flash 0x0 esp32_bootloader.bin
esptool.py v2.8
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting.....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b4:e6:2d:eb:1b:c9
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 65536 bytes to 12827...
Wrote 65536 bytes (12827 compressed) at 0x00000000 in 1.1 seconds (effective 461.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
(esp32) :~/workspace/python/esp32$ esptool.py write_flash 0x10000 WLED_0.11.1_ESP32.bin 
esptool.py v2.8
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: b4:e6:2d:eb:1b:c9
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1170864 bytes to 687308...
Wrote 1170864 bytes (687308 compressed) at 0x00010000 in 60.8 seconds (effective 154.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`
leads to the following outout on serial

`
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Ada
`

**Expected behavior**
Just to make it boot, so I can connect via WiFi to configure the WLED

**WLED version**
 - Board: ESP32_Devkitc_V4
 - Version 0.11.1
 - Format binary

**Additional context**
My setup it self is working, flashing micropython to same esp32 board works as expected and everything's running afterwards.

Thank you for your help!

Aircoookie, 2021-02-17:
Hi!
The line `Ada` in the serial output indicates that it was flashed and booted successfully. You should 1) see 30 LEDs turn on orange if attached to pin D4/gpio2 and 2) be able to connect to the `WLED-AP` WiFi network. Does that work?

rlfnb, 2021-02-17:
My bad! Yes its working as expected! Maybe a short "WLED started" would be good as well?

rlfnb, 2021-02-17:
I'll close as its for sure no bug at all :)

closed 2021-02-17 by rlfnb