wled/WLED #1761: Can't build platformio.ini

denisTenis1, 2021-02-17:
**Describe the bug**
cannot successfully build platformio.ini:

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.6.2) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.7 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting wled00.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Library Manager: Installing git+https://github.com/lorol/LITTLEFS.git
FileNotFoundError: [WinError 2] The system cannot find the file specified:
  File "C:\Users\Kristina\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Kristina\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Kristina\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Kristina\.platformio\platforms\espressif8266\builder\main.py", line 238:
    target_elf = env.BuildProgram()
  File "C:\Users\Kristina\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Kristina\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Kristina\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Kristina\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Kristina\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Kristina\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1062:
    project.install_dependencies()
  File "C:\Users\Kristina\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 898:
    lm.install(spec)
  File "c:\users\kristina\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "c:\users\kristina\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 86:
    force=force,
  File "c:\users\kristina\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\kristina\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\kristina\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55:
    src_dir, remote_url, tag, silent
  File "c:\users\kristina\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137:
    self.configure()
  File "c:\users\kristina\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146:
    result = proc.exec_command([cls.command, "--exec-path"])
  File "c:\users\kristina\.platformio\penv\lib\site-packages\platformio\proc.py", line 117:
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\Kristina\.platformio\python3\lib\subprocess.py", line 800:
    restore_signals, start_new_session)
  File "C:\Users\Kristina\.platformio\python3\lib\subprocess.py", line 1207:
    startupinfo)
===================================================================================== [FAILED] Took 1.84 seconds =====================================================================================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      FAILED    00:00:01.837
================================================================================ 1 failed, 0 succeeded in 00:00:01.837 ================================================================================
The terminal process "C:\Users\Kristina\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

**To Reproduce**
Steps to reproduce the behaviour, if consistently possible
Follow wiki steps: https://github.com/Aircoookie/WLED/wiki/Compiling-WLED

**Expected behaviour**
A clear and concise description of what you expected to happen.

To build successfully and upload it to my nodeMCUV3

**WLED version**
 - Board:
 d1_mini
 - Version
 0.11.1
 - Format
 Binary

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

gsieben, 2021-02-17:
use... 

default_envs = d1_mini

and it should work...

gsieben, 2021-02-17:
Ich selbst habe alle default_envs deaktiviert und die platformio_override.ini Option gewählt, um eigene Settings zu machen:

File: platformio_override.ini
# Example PlatformIO Project Configuration Override
# ------------------------------------------------------------------------------
# Copy to platformio_override.ini to activate overrides
# ------------------------------------------------------------------------------
# Please visit documentation: https://docs.platformio.org/page/projectconf.html

[platformio]
 default_envs = d1_mini
; default_envs = d1_mini_debug      // DOES NOT WORK!
; default_envs = d1_mini_ota

[env]
board_build.f_cpu = 160000000L              ; set frequency to 160MHz (if not set - it will be 80 MHz)
board_build.f_flash = 80000000L             ; set frequency to 80MHz
upload_speed = 921600 

; ####### Usermod: Temperatur Extension
[env:d1_mini]
build_flags = ${common.build_flags_esp8266} 
    -Wl,-Teagle.flash.4m2m.ld ; 2m SPIFFS & 1M OTA - In my opinion this is the best setting for the D1 Mini with 4 MB flash memory.

[env:d1_mini_ota]
build_flags = ${common.build_flags_esp8266} 
    -Wl,-Teagle.flash.4m2m.ld ; 2m SPIFFS & 1M OTA - In my opinion this is the best setting for the D1 Mini with 4 MB flash memory.
 
; upload_port = "10.7.7.25"    # WC Boden
 upload_port = "10.7.7.4"     # WC Decke
; upload_port = "10.7.7.34"    # Lucas Zimmer
; upload_port = "10.7.7.x"     # Bad

upload_flags =
    --debug

denisTenis1, 2021-02-17:
> use...
> 
> default_envs = d1_mini
> 
> and it should work...

Tried using a lot of environments with no success...

denisTenis1, 2021-02-17:
> I myself disabled all default_envs and selected the platformio_override.ini option to make custom settings:
> 
> ```
> File: platformio_override.ini
> # Example PlatformIO Project Configuration Override
> # ------------------------------------------------------------------------------
> # Copy to platformio_override.ini to activate overrides
> # ------------------------------------------------------------------------------
> # Please visit documentation: https://docs.platformio.org/page/projectconf.html
> 
> [platformio]
>  default_envs = d1_mini
> ; default_envs = d1_mini_debug      // DOES NOT WORK!
> ; default_envs = d1_mini_ota
> 
> 
> [env]
> board_build.f_cpu = 160000000L              ; set frequency to 160MHz (if not set - it will be 80 MHz)
> board_build.f_flash = 80000000L             ; set frequency to 80MHz
> upload_speed = 921600 
> 
> ; ####### Usermod: Temperatur Extension
> [env:d1_mini]
> build_flags = ${common.build_flags_esp8266} 
>     -Wl,-Teagle.flash.4m2m.ld ; 2m SPIFFS & 1M OTA - In my opinion this is the best setting for the D1 Mini with 4 MB flash memory.
> 
> [env:d1_mini_ota]
> build_flags = ${common.build_flags_esp8266} 
>     -Wl,-Teagle.flash.4m2m.ld ; 2m SPIFFS & 1M OTA - In my opinion this is the best setting for the D1 Mini with 4 MB flash memory.
>  
> ; upload_port = "10.7.7.25"    # WC Boden
>  upload_port = "10.7.7.4"     # WC Decke
> ; upload_port = "10.7.7.34"    # Lucas Zimmer
> ; upload_port = "10.7.7.x"     # Bad
> 
> upload_flags =
>     --debug
> ```

Followed your steps, but got the same error, tried reinstalling PlatformIO with no success

Doyle4, 2021-02-17:
Try downloading Visual Studio Code, I use that mainly for building, I also use custom_LEDPIN_4

Using Arduino I use: Board: WeMos D1 R1, Programmer: AVRISP MKII <-- This also works for me with NodeMCU ESP8266 boards.

denisTenis1, 2021-02-18:
> Try downloading Visual Studio Code, I use that mainly for building, I also use custom_LEDPIN_4
> 
> Using Arduino I use: Board: WeMos D1 R1, Programmer: AVRISP MKII <-- This also works for me with NodeMCU ESP8266 boards.

I'm using VS code. The problem isn't with the board selection, I went through all the boards and got the same errors while building

Doyle4, 2021-02-18:
[image]

Screenshot of my VCS for D1 Mini, after changing environment, make sure to press the trash a couple times (im sure you are aware of this) if this doesn't help, closing VSC and reopening usually does the trick for me.

J0RD4N300, 2021-02-19:
I had the same issue.

This line:
Library Manager: Installing git+https://github.com/lorol/LITTLEFS.git

Kept appearing as Git wasn't configured properly in VS Code.

Double check Git is installed/configured properly and see if that works.

d3sbo, 2021-02-26:
on windows 10 I had to get python from the microsoft store just could not get it working from the python installer even if I pointed vs code to the python install folder

jfcabral, 2021-05-26:
This error has just happened to me. As @J0RD4N300 said it was related to git, as I didn't have the `git.exe` configured in my `PATH` environment variables.

closed 2022-04-16 by stale[bot]