wled/WLED #1768: UDP Realtime Leaves an Off Relay Turned Off

jdtsmith, 2021-02-22:
**Describe the bug**

When sending RealTime UDP data, it's very convenient that whatever WLED is doing gets interrupted to display the data stream, and then after that ends, WLED goes back to whatever it was doing before.  But this doesn't work when using a relay which is off.  

If the strip is "off" (aka zero brightness), receiving new realtime data leaves it off, and therefore `handleIO` never turns the relay on.  

**To Reproduce**

Enable and connect a relay, and send UDP realtime data to the strip when it is "off".

**Expected behavior**

The strip and relay turn on when receiving realtime data, then turn back to the state they had before receiving (including "off"!).

**WLED version**
 - Board: ESP32 dev
 - Version 0.11.1
 - Format self-compiled

**Additional context**

A good solution might be as follows:

```diff
modified   wled00/udp.cpp
@@ -83,6 +83,11 @@ void realtimeLock(uint32_t timeoutMs, byte md)
 
   realtimeTimeout = millis() + timeoutMs;
   if (timeoutMs == 255001 || timeoutMs == 65000) realtimeTimeout = UINT32_MAX;
+
+  if (bri == 0 && realtimeMode == REALTIME_MODE_INACTIVE && md != realtimeMode) {
+    strip.setBrightness(scaledBri(255));
+  }
+
   realtimeMode = md;
 
   if (arlsForceMaxBri && !realtimeOverride) strip.setBrightness(scaledBri(255));
```

I've tested this and it works pretty well, though I noticed the brightness control actually operates to scale down the brightness of UDP realtime data when a stream starts while the strip is on, whereas this code cranks brightness all the way up.  Not sure which behavior I prefer!

blazoncek, 2021-03-18:
A better approach is to use toggleOnOff() instead of strip.setBrightness(scaledBri(255));

A question remains if you want to switch relay and LEDs off after realtime data is done or not.

jdtsmith, 2021-03-19:
Thanks for the idea to use `briLast`.  You can't use `toggleOnOff` because the UDP code sets the strip back to `bri` after realtime streams stop.  I added a [similar fix](https://github.com/Aircoookie/WLED/blob/1f871690655a280c4a74bc681f7c0cb2d4d79247/wled00/udp.cpp#L93) in my recently submitted Realtime Smoothing PR (#1808). 

It's actually very nice to have it come on with the last brightness level when realtime streaming starts.  You can always turn on `arlsForceMaxBri` if you want it to be at max bright for realtime (though I'm not clear how to toggle that in the interface).

blazoncek, 2021-03-19:
> (though I'm not clear how to toggle that in the interface).

You can't. There is no API to do that.
It is just a Settings override (permamnent until settings change).

Aircoookie, 2021-03-20:
Hi and thank you very much for the discussion and proposal!
Let me summarize: If realtime data is sent while the strip is off, no LED signal is outputted _and_ the relay (if connected) stays off, unless `arlsForceMaxBri` is selected.

In that case, it is not an issue with relay operation itself, but with the LEDs not turning on in general (even when permanently powered if no relay is used).

It is intended behavior that the set master brightness and on/off state apply to a realtime signal (with the exception of the `arlsForceMaxBri` override). Therefore, lights will stay off if they were off, even in the event realtime is received.

That said, turning on the lights to the last brightness on start of realtime mode might be useful too depending on use case. The only potential issue I see is a user turning off the light while receiving realtime data and it turning back on inadvertedly due to a brief interruption of the stream. How about we change the `arlsForceMaxBri` setting to be "Realtime brightness" and have three values: "Maintain" (current unchecked behavior), "Last" (this proposal, only differs from "Maintain" if turned off in that it will turn on to last brightness for the duration of the realtime stream), and "Maximum" (current checked behavior)?

jdtsmith, 2021-03-20:
Your summary is perfect.  The relay is as you say not the real issue; it was rather my expectation that a realtime stream would "turn on" the strip if it was off when the stream started.  

> How about we change the arlsForceMaxBri setting to be "Realtime brightness" and have three values: "Maintain" (current unchecked behavior), "Last" (this proposal, only differs from "Maintain" if [...]

Thanks, this sounds like a great solution.  I suppose in some situations, if realtime data come in unexpectedly, it might be alarming if the lights turn on.  For me, I'd like to send a sound-reactive UDP stream and have the lights turn on whether or not they were on beforehand (then turn off afterwards).  In my playing around with the fix in #1808, I actually enjoy being able to scale the realtime stream's brightness up and down.  Do you anticipate a run-time UI setting for Realtime Brightness?  BTW, it would likely make sense to have _all_ the different ways data can be sent in to WLED respect this setting.  It's not entirely clear if Hyperion or ES131 or TPM2.net are considered "realtime"; probably more of a documentation fix.

Current dev resets brightness to 255 on each and every realtime packet if arlsForceMaxBri is true.  But I suspect even with "Maximum", the user would like to be able to change brightness if that's too bright, and have it persist for the duration of the current stream.

blazoncek, 2023-05-26:
This has been fixed in https://github.com/Aircoookie/WLED/commit/46b66c76ef5ce36e7dc2d2f2dbafa8036ddf3d4c

closed 2023-05-26 by blazoncek