wled/WLED #1774: WIFI-AP only visible for few seconds then goes away on nodemcu v2 lua boards

koenslootmans, 2021-02-24:
Hello,

This is my second board I'm trying to configure and I experienced the following problem with my first board:

board: Nodemcu lua v2
WIFI-AP is accessible for just a few seconds, then goes offline (using latest version WLED_0.11.1_ESP8266)
If you  manage to connect and setup the wled configuration then the wled device constantly goes off-line and you can not connect until rebooted

Disabling the wifi power saving settings fixes the problem

--> I am now trying to set up a second board but I cannot, The wifi-ap does not stay visible long enough. 
I'm new at using arduino boards 

Is there a way that I could compile my own wled version but with the wifi power saving disabled by default, I believe this would solve my setup/wifi-access problem?

Any help would be appreciated (I have done the OTA flashing method using the arduino ide but the result was the same...)

Thanks,
Koen

koenslootmans, 2021-02-24:
I have been able to compile using vscode and found the settings for disabling the wifi sleep
However no luck, the behaviour is the same

I guess the problem is the board, could someone guide me to which boards work well 
Unless there are some settings I can change...

sansillusion, 2021-02-24:
Are you sure your power supply if strong and stable enough. ESP's are sensitive to electrical noise and power stability.

If you can try a different power source, if using usb cable try a different cable or use external power.

koenslootmans, 2021-02-27:
Hello,
I have tried but it does not work,
I gave up and got a D1 mini and that works perfectly

Thanks for the reply

sergge1, 2021-03-07:
I have slot is setups based on nodemcu v2 (cp2102), works good. Unstable access point often caused by not good dc supply.

closed 2021-07-15 by stale[bot]