wled/WLED #1777: Power On does not work if WS2812B connected

DSRocker, 2021-02-26:
Everything is connected correctly. If the LED strip is connected to the ESP GPIO2 and I power on the esp, then it does not work. Only the first LED is on. If the strip is disconnected and I power on the esp and then connect the strip, then it works fine.

tonyn0, 2021-02-27:
This likely means you need a pull-up resistor (~10k) between the gpio and 3.3v pins.

DSRocker, 2021-02-28:
Which GPIO?

DSRocker, 2021-02-28:
I tried it with GPIO 0 and GPIO 2. It does not work

pa1okz, 2021-03-01:
R U sure you use the same GND for the LED strip and the ESP? Make sure that no excessive current from the LED strip is drained from the ESP PCB. Here is how connections shall be done: https://github.com/Aircoookie/WLED/wiki

bl33ding-edge, 2021-03-10:
I had the same problem, the pull up resistor on gpio 2 did the trick!
3V3---10K---D2---WS2812B.DIN

ashinckley, 2021-03-16:
I've tried a pull up resistor of 10k as suggested in this thread, a 3.3k and a 4.2k as suggested in the [FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ#once-i-attach-leds-i-cant-connect-to-the-esp-anymore).  Each time I power on with the D2-DIN connection, resistor or not, I cannot see the ESP WiFi AP.

DSRocker, 2021-03-16:
You have to use another GPIO! I think there is no other way.

  (#1777)

ashinckley, 2021-03-16:
Which GPIO? Were you able to get it to work?  Did you have to recompile with a different GPIO value?

DSRocker, 2021-03-16:
Yes... it works with gpio 14... you have to recompile with other gpio

closed 2021-07-22 by stale[bot]