wled/WLED #1778: Change ESP8266_1M Pin and 5050 support.

ZeddBlock, 2021-02-26:
Hi ! I really love your work, thank you so much. Forgive my english.
But, I have two requests, for 2 news .bin files, which concerns the "WLED_0.11.1_ESP8266_1M.bin".

- Firstly, with the ESP01, I can't boot when the GPIO2 is connected to the led. I need to unplug it, boot it, and then reconnect LEDs and it works. Is it possible to upload the same bin file but with one other GPIO used for Led Data instead of GPIO2 ?

- And, is it possible to upload the same bin file, but for 5050 non adressable led strips with GPIO0, 1 and 3 ? I'll use Mosfet.

Thank you so much for your work, and your help.

DSRocker, 2021-02-28:
Hi. I have the samw issue. I don't know the solution. If you have it, please tell me the way to fix it.

ZeddBlock, 2021-02-28:
> Hi. I have the same issue. I don't know the solution. If you have it, please tell me the way to fix it.

I found the solution to change the GPIO on the esp01. 
I just downloaded the wled00 folder, opened the Arduino project, and changed the led pin (2 -> 4). 
I send it the my esp01, and everything works fine !

DSRocker, 2021-02-28:
In which file did you change the GPIO?

ZeddBlock, 2021-02-28:
In "NpbWrapper_h"

ddijanic, 2021-02-28:
In "NpbWrapper_h"
You cant set pin for non adressable LED... 
//#define WLED_USE_ANALOG_LEDS

blazoncek, 2021-03-17:
@ZeddBlock @DSRocker you may need a pullup resistor on gpio2 (the default pin).
If you want to freely change pins (during runtime) you can use v0.12 (in beta state, but works ok ATM).
Checkout the dev branch and compile yourself or use precompiled versions by @srg74 in his WLED_Wemos_shield.
There is also a stable multipin 0.11.1 version by me (also possible to change pins during runtime).

closed 2022-07-25 by blazoncek