wled/WLED #1787: Is itpossible to ignore a broken led in a WS2812B strip?

o-s-e, 2021-03-02:
Hi,
I use homeassistant to control all WLED instances. (I have 3 esp32 controlling 3 different strips)

I have an issue with  a ~7m strip of WS2812B in my RV.
A few days ago, the strip started "glitching". Part of the strip works fine, the other one blinks, shows different colors,..
I was able to nail down the issue to led #204 in the strip.
To test my hypothesis, I created 2 segments: led#1 - led #203 and led #205 - led #376.
Now both segments are working fine, but it makes it a mess to control.
The wled integration in Homeassistant creates switches per segment and now I'm forced to control them separately.

My questions:

- Is there a way to exclude 1 led?
- If not, is there a way to "combine" segments? 

I don't want to replace the whole strip, because it's quite inaccessible mounted in a cove/shadow line.

Thank you.

ZBlackw1, 2021-03-02:
You should be able to accomplish this by remapping the leds to make the broken LED be at the end. In your case, the LEDs would be mapped in the order: LEDs 0 to 203, then LED 205 to the end of the strip, then LED 204. Then you just set up the segment with one less than the total number of leds you have. Here is the wiki article for LED Mapping that shows you how to do it: [https://github.com/Aircoookie/WLED/wiki/Mapping](https://github.com/Aircoookie/WLED/wiki/Mapping)

o-s-e, 2021-03-02:
Thank you ZBlackw1, that worked perfectly.

closed 2021-03-02 by o-s-e