wled/WLED #1791: DHCP TTL problem

dwillmore, 2021-03-08:
I'm noticing that my WLED board (ESP8266 ESP-01 w/1MB) is behaving strangely WRT DHCP.  I recently replanned my network--moving to have my dynamic address, etc.  As part of that work, I set the TTL for DHCP to 600s.  Once all the work was done, I moved it back up to the normal 86400s value.  Everyone picked up new leases and have been behaving themselves except for the WLED board.  It got a new lease, but it seems that it thinks its TTL is still 600s as it's requesting a new lease every 300s (50% of TTL).  The DHCP server outputs a little log entry every time this happens.

I have reset the WLED board and that made no change to its behavior--it seems the wrong TTL value is persistant through reboots.  I have verified the lease on the server and it looks sane.

This may be a little difficult to reproduce, but if you follow my setup, it may work for you.  I'm not sure exactly how we got into this state, so I don't know if I would reproduce it.

I would have expected WLED to pick up a new lease and respect the changed TTL value.

My hardware is a horrible lashup of an ESP-01 w/1MB running the 1MB ESP8266 binary from the v0.11.1 release Build 2012180.

I am guessing this is actually a problem with the underlying networking library, but it may be that WLED is doing something unusual.

WLED has been a very useful project for me and I would like to thank all of the developers for their work!

Aircoookie, 2021-03-08:
Hi, thank you very much for reporting!
That is indeed very strange, likely a bug in the WiFi/DHCP handling part of the ESP8266 Arduino core.
It seems like here: https://github.com/esp8266/Arduino/issues/7020 both a Tasmota and a fellow WLED user had the same or a similar problem.

dwillmore, 2021-03-08:
I managed to 'fix' it by putting in a static allocation in the WLED GUI (same as the assigned address) and power cycling the device--it dropped off the network when I went to the static allocation, but came back after the power cycle.  I then cleared out the static info so that it would go back to DHCP.

At that point, it did a DISCOVER and the server got grumpy and complained that it already had a lease, so why was it asking, but here take this old lease that's still valid.  From that point on, the WLED device picked up the new TTL and hasn't been pestering the server.  So, that's a 'fix'.

I'll close the issue here and head over to the Arduino issue to persue a fix there.  Thank you for your time!

closed 2021-03-08 by dwillmore