wled/WLED #1792: Unable to connect to WLAN

kanawati975, 2021-03-09:
Greetings everyone, and thanks to everyone helped in this great Project.

I have an issue and I was hoping that someone could tell me what am I doing wrong.

So... Using Platformio I have uploaded the code to a D1 mini, which was without any Problems... Accessed WLED-AP and assigned wifi data, and saved. But it won't connect to the wireless network. (I'm also checking my Router for any connection).

Tried fiddling around with the code, and uploaded it again, to no avail.

Tried another D1 mini Chip... disabled unneded functions... also didn't help.
Tried accessing the D1 mini via 4.3.2.1 and re-flashed using the .bin file... also didn't help
Tried deleting the code, and uploading on a trusted NodemcuV2 but also didn't work.

What am I missing? Why can't it connect to wifi?

Any help would be appreciated.

gobo-ws, 2021-03-09:
Now I do not know if it is due to this but do you use any special characters in your WPA2 key?
If so, they need to be escaped.
https://github.com/xoseperez/espurna/issues/1439

kanawati975, 2021-03-09:
@gobo-ws Actually there is one. But not sure if that's somehow relevant

closed 2021-07-15 by stale[bot]