wled/WLED #1793: Unable to connect to WIFI

LeducH, 2021-03-09:
**Describe the bug**
I am unable to connect to my home wifi 

**To Reproduce**
connect to the via the wifi setup 
enter SSID and password 
it shows client ip : not connected 

**Expected behavior**
connects to my home wifi 

**WLED version**
 - Board: ESP8266 node mcu
 - Version 0.11.1
 - Format `WLED_0.11.1_ESP8266.bin`

**Additional context**

Aircoookie, 2021-03-09:
Hi, sorry about that! Is your WiFi 2.4GHz? What router/access point do you use?

LeducH, 2021-03-10:
I use a fritzbox with 2.4 and 5GHz connection. But it is connected via 2.4Ghz

sigsegv0, 2021-03-18:
How about the case you're using for SSID? I want to say it's case sensitive and needs to  match the router's definition of SSID.

closed 2021-07-23 by stale[bot]

rjworks, 2022-07-27:
did you fix it