wled/WLED #1798: WLED 12v

fm20000, 2021-03-12:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

JoshWobbles, 2021-03-13:
WLED has nothing to do with the voltage you use.

fm20000, 2021-03-14:
I found what I was looking for regarding the question in my post but thanks. 😂

closed 2021-03-19 by Aircoookie