wled/WLED #1799: Esp32 Multi Strip /io

9ghost9, 2021-03-12:
Having disabled Crossfade in WLED only seems to affect the native Gpio and not the added Strips/Gpio .

Any help suggestions would be greatly appreciated .

Also , thanks so much for adding this mod .

blazoncek, 2021-03-15:
If you are willing try the beta version (0.12).
You can get support on WLED Discord.

gabyleon, 2021-03-16:
excuseme , where is the beta version?

blazoncek, 2021-03-16:
> excuseme , where is the beta version?

You can compile it yourself from dev branch or use a precompiled version available at @srg74 WLED_wemos_shield repo.
https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/WLED_wemos_shield

Please join WLED Discord for further support.

gabyleon, 2021-03-16:
@blazoncek, thank you, sir :-)

9ghost9, 2021-03-16:
> If you are willing try the beta version (0.12).
> You can get support on WLED Discord.

You are awesome my friend !!! thank you very much for this works like a charm.

closed 2021-03-16 by 9ghost9