wled/WLED #1801: [Feature Request] Ability to send effect for a said duration (like a notification)

insane2subro, 2021-03-13:
Hi, I was wondering if we can send a request via API or MQTT where an effect is played only for a duration as specified. This will be great in showing phone or home assistant notifications.

Use case: Say someone is at the main door and I missed the homeassistant notification(pir) but if the leds could flash in a said color (Red) then it would be a visual notification.

Thanks a lot :)

sansillusion, 2021-03-13:
This can all be achieved in HA itself.

This is my script in HA for random effects on two wled devices. You can
extrapolate what you need here I think.

repeat:
  sequence:
    - data_template:
        rgb_color:
          - '{{ (range(0, 255)|random) }}'
          - '{{ (range(0, 255)|random) }}'
          - '{{ (range(0, 255)|random) }}'
      entity_id: light.technics
      service: light.turn_on
    - data_template:
        effect: '{{ (range(0, 115)|random) }}'
        intensity: '{{ (range(50, 255)|random) }}'
        speed: '{{ (range(50, 255)|random) }}'
      entity_id: light.technics
      service: wled.effect
    - data_template:
        rgb_color:
          - '{{ (range(0, 255)|random) }}'
          - '{{ (range(0, 255)|random) }}'
          - '{{ (range(0, 255)|random) }}'
      entity_id: light.tv
      service: light.turn_on
    - data_template:
        effect: '{{ (range(0, 115)|random) }}'
        intensity: '{{ (range(50, 255)|random) }}'
        speed: '{{ (range(50, 255)|random) }}'
      entity_id: light.tv
      service: wled.effect
    - delay: '45'
  while:
    - condition: state
      entity_id: input_boolean.random_effect
      state: 'on'

insane2subro, 2021-03-14:
Yes, too am doing something similar, but with this, the problem is that WLED won't resume back to its old state. Say for example its on with a single color, this will change it and needs to be manually changed back.

closed 2021-07-21 by stale[bot]