wled/WLED #1802: Wemos D1 mini Pro

Rafal-PL, 2021-03-13:
**Describe the bug**
A clear and concise description of what the bug is. Please quickly search existing issues first!

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

srg74, 2021-03-14:
Please visit Discord WLED channel for further issue investigation

Sangoku, 2021-03-15:
I am using an esp32 D1 mini custom compiled and WS2812B  works without problems. Could you try to add the data pin after some length? 
It could be your 1st led is broken.

Rafal-PL, 2021-03-15:
Hello everyone, I have a wemos d1 mini pro and a led string
[image]

I connect: Pin D4  - Data, power pin 5V and GND and it does not work. I connected the power supply to the 3.3V pin once and it worked for a while and stopped for a while. I connect the same LEDs to ESP32 and everything works. Somewhere read that you may need to apply a resistor on pin D4?

closed 2021-07-30 by stale[bot]

LocalNic, 2021-12-02:
Looks like the same or similar problem that I have faced and documented my findings here #2342

blazoncek, 2021-12-02:
These may not be SK6812 but are WS2812 instead.

darkmattercoder, 2022-02-21:
I experience the same, with the same product. They seem to have a white led, so they cannot be ws2812 afaik. I also have very weird behaviour. Data line looks like being not in the middle but on the outside. However, what weird is, even if I connect the outside line to data from the wrong end, controlling kinda works (when set to sk6812, used a lot more leds than there physically are) however colors are messed up completely. If I use the middle wire for data, no matter which end i connect it, no controlling works at all. Neither with ws281x nor sk6812 settings. 

Atm I am lost. A friend of mine apparently could control the strip with an sp110e controller. I still need to confirm that myself.

blazoncek, 2022-02-21:
Those may have addresses _burned in_ so they will still work even if one fails (bus topology as opposed to daisy chain topology).

darkmattercoder, 2022-02-21:
after further investigation in #2342 i can confirm what you are saying. References can be found in #1312. So it is no sk6812 at all. The white LED I did mentioned before does not exist in those. Thus is more like ws281x protocol-wise.