wled/WLED #1803: Request feature to select start and stop LED for hour and minutes

mattbruman, 2021-03-14:
I am building a clock but am going to be using two different sized strips with different amount of leds per meter. The lengths for the strips are almost exactly the same 12 on one and 60 on the other. I want to be able to tell wled that 1-12 is hours and 13-73 are minutes. This should solve the issue people have with not being able to tell hours from minutes. Would this be possibal as a feature? I'm willing to try beta code if it can be done. Ill be using a d1 mini.

Legsmaniac, 2021-03-14:
That would be really easy to do using a custom usermod coded file but unless you are planning on using any of the built in WLED effects, using WLED is a bit overkill for such a simple use.  I've seen there are already plenty scripts out there for simple clocks such as your idea, it's probably far easier to use those instead I would think.

mattbruman, 2021-03-14:
> That would be really easy to do using a custom usermod coded file but unless you are planning on using any of the built in WLED effects, using WLED is a bit overkill for such a simple use. I've seen [...]

Kinda liked the simplicity of using wled, easy to flash and keep updated with security patches and such. Do you happen to know where I can find any of those other projects? I'm not a programmer and don't really have time to learn which is why I wanted to use wled and since it already has a clock feature there's no reason to not ask for a more feature rich option.

Legsmaniac, 2021-03-14:
> no reason to not ask for a more feature rich option

Indeed.  Since what you're after is currently not implemented (as far as I'm aware) I just thought a specifically designed clock would be the order of the day, until and unless Aircookie does decide to add in what you're asking.  If you Google "WS2812 clock", you should find loads of projects.  Plenty on the likes of YouTube and Instructables.  I built my own Binary Clock using WS2812 LEDs and had to resort to alternative coding since WLED doesn't provide this and I'm very pleased with the results. 
Anyway, was just a thought.  Seriously, it's worth a Google as there's loads of fab and various designs out there.  😊

closed 2021-07-21 by stale[bot]