wled/WLED #1805: WLEDMasterLight object has no attribute 'async_preset' error in Home Assistant when calling service

xfader2018, 2021-03-17:
Hello,

While using WLED with multiple segments configured I noticed an error when calling the WLED service to activate a saved preset.

WLEDMasterLight object has no attribute 'async_preset'

This error does not occur when there are no segments. The call service does work though but it's returning this error.

Tested on multiple wled devices.

Here some more error details:

`Logger: homeassistant.components.websocket_api.http.connection
Source: helpers/service.py:673 
Integration: Home Assistant WebSocket API (documentation, issues) 
First occurred: 16 March 2021, 21:31:53 (3 occurrences) 
Last logged: 16 March 2021, 21:46:57

[140299175810240] 'WLEDMasterLight' object has no attribute 'async_preset'
[140299171967088] 'WLEDMasterLight' object has no attribute 'async_preset'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 141, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1488, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1523, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 578, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 642, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 681, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 673, in _handle_entity_call
    result = hass.async_run_job(partial(getattr(entity, func), **data))  # type: ignore
AttributeError: 'WLEDMasterLight' object has no attribute 'async_preset'`

Aircoookie, 2021-03-19:
Hi and thank you for the report!
What version of HA are you using?
As far as I know, the WLED integration creates a Master entity only when multiple segments are configured, if the light has just one segment, that is also used for the master brightness + on/off.
`async_preset` is not a key or command in the WLED JSON API, so it must be something internal to the integration.

@frenck do you know what happens here?

frenck, 2021-03-19:
Yep! That is a bug! Thanks @xfader2018!

However, could you please raise it at the Home Assistant issue tracker? As this is an bug in HA, not WLED.

https://github.com/home-assistant/core/issues

Thanks! 👍

xfader2018, 2021-03-19:
Thank you both for reply and confirming.
I was not sure wether it was a WLED or a HA bug so I started it to report here.
Will raise the issue at the HA tracker now.

Thanks!

closed 2021-07-30 by stale[bot]