wled/WLED #1809: Merry Christmas

Legsmaniac, 2021-03-19:
Just tried and had a play with the latest new dev version.
Wow!  Coming along nicely!  👍

But......

What's happened to Merry Christmas ?!?!! 🤔
All my Christmas display lights use this as a saved startup preset.
Please, please tell me this is just an oversight or simply not been added yet since this is a development version?
I LIKE Merry Christmas effect.   🎄🎁🎈🎅🤶
Should I flash my WLEDs to this latest upgrade in the future, what happens to my saved presets, particularly Merry Christmas?
Presumably they'll change to whatever number effect Merry Christmas was but is now?
This would mess up all my Christmas lights.  😒

By the way, my Nodemcu 8266 was already flashed with an earlier dev which I upgraded to this one via manual OTA update.
It had already been setup with a static IP address to connect to.  Normally, this IP is still there are upgrade but this time it was wiped and I had to set it up all over again through AP connection.  Was this intentional or a bug?

shauneccles, 2021-03-19:
Merry Christmas is just Running 2 with Red/Green, you can that as a preset instead.

The effects list is getting trimmed because a lot of them are just redundant.

I can't speak for what will happen to the effect presets when they change.

blazoncek, 2021-03-19:
Exactly.

brianmiller1956, 2021-03-30:
I too was surprised to find Merry Christmas had been removed, as you describe it is easy to replace it with running 2, however a new user to WLED may look at the available effects and think "no Christmas effects". This can't be much good.
Halloween and candy cane could also be removed as they are effectively the same.

Legsmaniac, 2021-03-30:
> and think "no Christmas effects". This can't be much good.

Indeed. Christmas is one of the main uses for WLED and often the reason such a device is searched for.  I may stick to and leave my current Christmas displays on v11
There's the old saying "If it's not broke....." but in this case would end "... don't upgrade."  I'm going to have to make sure I keep copies of the older sources just in case they ever disappear for download. 
I think it's wonderful people are getting their requests added into WLED but I can't help being a little concerned for the future versions as I feel they are starting to get more and more complex and "bogged down". A bit like Microsoft Windows!
It would be cool if many if the extra features could be available as plug -ins whereby people could add only the features they need to a basic WLED module. 
Hey, I can dream.....

colino17, 2021-07-06:
I'm not opposed to reducing redundancies, but as devil's advocate here, a lot of WLED users, including myself, use it through the Home Assistant integration. Currently there does not seem to be a way to natively specify secondary/tertiary colors for effects in Home Assistant. This means that there is no easy way to set "Running 2 with Red/Green" within Home Assistant itself.

Legsmaniac, 2021-07-06:
I use HA and set some routines using Automation and Scripts.  You can use any of or a mixture of Presets, Palettes, Restful and/or MQTT.  It's not difficult.

pepsonEL, 2021-11-19:
Is any chance to anybody describe me step by step how add Merry Christmas effect manually ? I never do it and please help...

blazoncek, 2021-11-19:
Select Running 2, then primary color (1) and Red, then secondary color (2) and Green.
If you want 3 colored Christmas select Tri Case effect, then primary (1) and Red, then secondary (2) and green, then tertiary (3) and White. Similar can be Rainbow and palette Colors only.
Play with sliders afterwards.

Legsmaniac, 2021-11-19:
> Is any chance to anybody describe me step by step how add Merry Christmas effect manually ?

@pepsonEL  What I did is changed the Halloween effect to Merry Christmas since I do not need or want Halloween so it was easier just to change it to Merry Christmas.  So all I did was auto-search through all the files for "Halloween" making sure I avoided "Halloween Eyes" and changed the wording to either "Merry Christmas", "merry_christmas" or "MERRY_CHRISTMAS" as appropriate, following the same convention for each one found and then changed the Halloween running effect colours from purple/yellow to red/green and then recompiled.  You could do the same for Candy Cane if preferred.  I wanted to keep Candy Cane, personally. 
It was easier and quicker than adding Merry Christmas in as a completely new effect.  😉🎅

fishbone-git, 2021-11-19:
this also works:
http://<wled_ip>/win&T=1&AC=255&CL=hFF0000&C2=h00FF00&C3=h000000&FX=37&SX=128&IX=128

the url can be changed to also alter speed (&sx=###) and intensity (&IX=###) to suit your needs.

pepsonEL, 2021-11-19:
Ok but i want to have predefined in WLED.... no by link....

fishbone-git, 2021-11-19:
yes, once you do any of the above methods, click create preset. Then you have an exact setting you can always call back later.

blazoncek, 2021-11-19:
... or you program the effect yourself as @Legsmaniac suggested.

pepsonEL, 2021-11-19:
Sorry i don't know how I can create it.... Sorry i am beginner user in wled

Jugrnot, 2021-12-18:
> this also works: http://<wled_ip>/win&T=1&AC=255&CL=hFF0000&C2=h00FF00&C3=h000000&FX=37&SX=128&IX=128
> 
> the url can be changed to also alter speed (&sx=###) and intensity (&IX=###) to suit your needs.

Trying to decode the "Running 2" which was **actually** "Chase 2" so I can get my Christmas Lights in a preset, your URL was unlimitedly helpful.  Thank you very much.

closed 2022-04-17 by Aircoookie