wled/WLED #1811: Unable to override preset through json api

keimpema, 2021-03-19:
**Describe the bug**
When a preset is active it is not possible to override it throught the api by setting ps to -1.

**To Reproduce**
1. set a preset either through the api or the ui
2. retrieve the state throught the api -> the ps attribute in the state object is set
3. set effect and palette through api, don't set preset (ps = -1)
4. retrieve the state throught the api -> the ps attribute in the state object is still set

[image]

**Expected behavior**
The preset should be reset when the individual properties are changed through the api -or- the preset should be allowed to be reset by providing the value -1 explicitly

**WLED version**
 - Board: wemos d1 mini esp8266
 - Version 0.11.1
 - Format Binary

blazoncek, 2021-03-19:
Presets start with 1, not with -1.
What you are trying to do is setting a preset number "-1" to a new values. Not resetting preset "3".
And by resetting a preset what do you mean by that? There are no "default" values for preset.
If you want to "remove" a preset there is a different API.

keimpema, 2021-03-19:
When no preset is set I receive ps=-1 in the GET json/state response. That's why I thought I had to use that value in the POST as well. But -1 is an illegal value. Since you mentioned presets start with 1 I just tried it with ps=0 and that works as expected. After that there no longer appears a preset active in the UI.

blazoncek, 2021-03-19:
You can close this issue.

closed 2021-03-19 by keimpema