wled/WLED #1814: NodeMCU ESP8266 hw-628 v1.1

stream74, 2021-03-21:
Hello

sorry for bad english, i'm french :)

I've got some problem with code

i try many thing but nothing work
it work perfectly with esp32 but not with esp8266 version hw-628 v1.1

i think i know what the problem but too many thing to change for me

when i change the   WiFi.begin(clientSSID, clientPass);
who are 
WLED_GLOBAL char char clientSSID[33] _INIT(CLIENT_SSID);
WLED_GLOBAL char char clientPass[65] _INIT(CLIENT_PASS);

by

const char *clientSSID= CLIENT_SSID;
const char *clientPass = CLIENT_PASS;

  WiFi.begin(clientSSID, clientPass );

it connect to my router
if i don't do these it doen't see my router

the same thing i think with the appass
i'cant connect to wled password doesn't work

i think there is a problem with recognize string

ghpk, 2021-03-23:
Are you compiling from available source,
If Yes, Why not use pre-compiled bin file Aircookie has shared already.

stream74, 2021-03-23:
hello
i try many thing
compil whith arduino
and pre-compiled fin
same thing happen

sansillusion, 2021-03-23:
I guess you have some special characters in your SSID or passkey. You may need to escape those characters. But standars don't recommend using special characters on an AP/router

stream74, 2021-03-23:
thank you for your answer
i know i don't speak very well english

he don't see my router
but the same thing happen when i wan't to connect to it with acces point
when i enter the _**AP_PASS     "wled1234"**_ 
he don't recognize it and disconnect

but i don't have this problem whith the example code esp8266 wifiacces point

wled code don't recognize any string

no problem with my esp32

only problem with the ESP8266 hw-628 v1.1

but when i change code like esp8266 example:
**const char *ssid**

it recognize everything

stream74, 2021-03-27:
hum i don't know why it don't work with my esp8266 version hw-628 v1.1
i test with esp 01S and no problem

closed 2021-08-03 by stale[bot]