wled/WLED #1817: how to setup few ESP32 in combination with segments?

EagleEi, 2021-03-22:
Hi All,
first thank you for this great OpenSource Project I love it!! 
I will do an light instalation in a small Forrest, based on the around 6pc  50x WS2811 wired 12V (christmas light), for example (https://youtu.be/xwxKnO8Ska0?t=220).
I will cut each 50x wire in 12WS811 Segments and place around 24pc of each 12x Segment side by side on Tree trunk.

Of Cause I will split it also with WLED in 12 Segments, and wire each DI to DO. 
In the description I understand that I can handle max. 15 Segemnts right now, thats a lot, but I´m afraid that the around 300 LEDs are not reliable enough, cause many segments, many meter Cabel and endless options for broke the Signal wire in this chain. 
So will build Sub-groups with ESP32 (wemosD1). based on this lets asume I will use  6x Wemos D1, each are 4 Segments with each 12LEDs. 
This is my Idea, and I don´t know how I can combine this Sub-groups of ESPs. 
My wish would be, for example, to set in WLED ESP1 left0m, ESP2 left 3m, ESP3 left 6m ....   so that the effect can turn in a circle. 
I tested right now two ESPs and I used the SYNC option and I´m wondering that some effects, like Fire or others ar in dependency, but actual I don´t know how to manage or how it is with 6 ESPs?

So I hope that my Forrest DIY Light show can go with this great tool!?
Thank you for your Ideas and this great Project!
br EagleEi

blazoncek, 2021-03-24:
You will not be able to have effects synchronised across 6 ESPs using just Sync feature.
There is currently no way to synchonise effects prperly (they will trigger at approximately same time but may diverge over time).

EagleEi, 2021-03-24:
Thank you Blaž for answer! Yes I´ve tested yesterday with 3 pcs of WemosD1 and for effects like strobe ther is sometimes an small offset, but very small. I think for the most "just" effects in combination with segments, the time offset between the ESPs is is ok, for the Idea for "running Letters" or animation from left to right probably not tight enough. Maybe I have to think about come over DMX ArtNet to give each ESP an own Number and maybe than the LEDs should be controlled by an other tool?!? ...  I´m just a beginner within this Project!

closed 2021-07-30 by stale[bot]