wled/WLED #1818: Latest 0.12.0b1 effects bug

Doyle4, 2021-03-22:
**Describe the bug**
Effects buttons no longer work

**To Reproduce**
Select any effect

**Expected behavior**
New effect to start

**WLED version**
 - Board: d1_mini
 - Version 0.12.b1 latest
 - Format self-compiled

Aircoookie, 2021-03-22:
Hi and thank you for reporting! That does not sound good.
Are you using the iPhone app, Safari or a different browser? What devices does it happen on? Do you have multiple segments configured? Will the new effect select in the UI (button becomes highlighted) or does nothing happen at all?

Doyle4, 2021-03-22:
Thanks for getting back to me.

I am using Google Chrome for iOS, but I can confirm it is also doing it on Safari on the iPhone also, I have cleared cache and data on both and effects still don't work, the button looks like it has been tapped, but does not get the highlight white dot.
I'm using 6 leds on a strip, it's my tester board, nothing special.

I can confirm that the effects do work on Google Chrome browser on my Macbook pro, so i'm guessing its the mobile version the bug is.

Doyle4, 2021-03-23:
Going to take a guess at something here.

I flashed over 0.12.b2, tested on my phone and everything worked, I presumed the new colour and effects layout had reverted back for mobile devices.
Connected my Macbook pro to the d1 mini, and saw the new UI for colour and effects, all worked great.
Reconnected back to my iPhone and the new UI was there this time, new colour preview/buttons and effects no longer worked, but did when back on my mac.

Could this be a Google sync issue? or a cache issue? I flashed the d1 mini back to 0.11.1 yesterday which gave me full control again on my iPhone, but as soon as my mac connects and I move over to my iPhone to use the new layout seems to be pushed over.

sansillusion, 2021-03-23:
I think it's just a browser cache issue. Had something similar happen. I
had to hard refresh chrome to get everything working like it should.

Doyle4, 2021-03-23:
> I think it's just a browser cache issue. Had something similar happen. I had to hard refresh chrome to get everything working like it should. Le mar. 23 mars 2021 4 h 15 a.m., Doyle4 ***@***.***> a écrit :
> […](#)

Think im going to stick with 0.11.1, I don't want to be clearing my iPhone cache each time sadly, plus the main project will be going to a first time user and last thing i want is constant calls ha.

sansillusion, 2021-03-23:
That is weird tho. Clearing cache should do it and it should keep working after. Unfortunately I don't have a iPhone to test. It works great on all my devices. Some needed a cache clearing but they don't revert to old UI on their own after.

Doyle4, 2021-03-23:
it only reverts back if I flash 0.11.1 which im fine with.
But if I flash the latest build 0.12.b2 or b1 which I have, the old UI shows on my iPhone, if I refresh the page the new UI then shows, I can select solid colours fine, but no effects or the drop down colour options work.

Very strange

Doyle4, 2021-03-23:
Cleared cache on my google and safari iOS browsers, still not able to change effects, defiantly an issue with new UI, something has to have changed.

Flashed back to 0.11.1, everything working great.

closed 2021-04-03 by Doyle4