wled/WLED #1820: Toggle in preset

aiskwad, 2021-03-23:
Hi) I apologize in advance for my English, it's the translator's fault)
When creating a preset, I specify **T=2 A=255 FX=0 CL=#ffeee4**, in which case the brightness parameter does not work. If I specify **T=2&A=255&FX=0&CL=#ffeee4**, then the brightness works, but toogle stops working, as if I specified **T=0**. If I specify **A=255 T=2 FX=0 CL=#ffeee4**, then the brightness works, but toogle works like **T=1**.

How can I make one preset set to always on / off with a given light and maximum brightness, no matter what I set before?

Wemos D1 mini, ESP-12, v0.11.1 Binary

Thanks in advance for solving my problem)

blazoncek, 2021-03-24:
Brightness (A) >0 also means "Turn on". Brightness (A) =0 also means "Turn off".
Brightness and toggle switch are mutually exclusive (you cannot use both in the same API call).

aiskwad, 2021-03-24:
> Brightness (A) >0 also means "Turn on". Brightness (A) =0 also means "Turn off".
> Brightness and toggle switch are mutually exclusive (you cannot use both in the same API call).

OK. can I turn on full brightness through the preset, but leave the function on off?

blazoncek, 2021-03-24:
I'm not sure I understand, but you can't do both at the same time.
You can use API (T=2) to toggle on/off without any other parameter.
And if you set brightness <>0 it will turn WLED on.

aiskwad, 2021-03-25:
> I'm not sure I understand, but you can't do both at the same time.
> You can use API (T=2) to toggle on/off without any other parameter.
> And if you set brightness <>0 it will turn WLED on.

I'm not sure I understand, but you can't do both at the same time.
You can use API (T=2) to toggle on/off without any other parameter.
And if you set brightness <>0 it will turn WLED on.

blazoncek, 2021-03-25:
> If someone used effects and left the brightness at <100%, I would like a short press on the button to turn on the white backlight at 100% brightness, and then turn it off with a short press.
That is a custom feature which is not implemented (using a button).
If you mean external source as a trigger (button) it is up to external source to store previous value and return wled to a state it was before the trigger happened. There is no such implementation in the code of WLED.

aiskwad, 2021-03-25:
> > That is a custom feature which is not implemented (using a button).
> > If you mean external source as a trigger (button) it is up to external source to store previous value and return wled to a state it was before the trigger happened. There is no such implementation in the code of WLED.

Okay thanks) I'll go edit the code)

aiskwad, 2021-03-25:
> Okay thanks) I'll go edit the code)

closed 2021-03-25 by aiskwad