wled/WLED #1822: [FR] deactivatable or favourable effects on the effect list

itCarl, 2021-03-24:
**Is your feature request related to a problem? Please describe.**
i really don't like how everytime i want a effect i have to search through he whole list of unwanted effects to search the one that i want. Yes, i could compile my own wled .bin but its very fiddly to make a new .bin for each an every wled client if you have a ton of these around the house. it would be way nicer to then set them up via the web api.

a **possible solution** would be to:
1. (in the Settings) be able to deactivate / activate effects depending on the if the user needs this effect or not
2. **OR** be ablte to favour a selection of effects that are than getting displayed at the very top of the effect list.

Legsmaniac, 2021-03-24:
**OR** use the Search Box atop the list of effects?

Just by typing in 2 or 3 letters shrinks the list of effects.

itCarl, 2021-03-24:
**hmm.. What Search box are u talking about?** 

---

[image]

Legsmaniac, 2021-03-24:
Hmmmm.   This one.......

[image]

itCarl, 2021-03-24:
thats weird... i dont have that.. which version are u running?

i use v11.1... isn't that the latest version?

---

[image]

itCarl, 2021-03-24:
i see... it just got added on March 18, 2021 https://github.com/Aircoookie/WLED/commit/4a20f43fbf5dab03295080f8b69ae0c448064588

But i still think that a search bar ist still to cumbersome (<= i'm from germany i just used a translater and i hope that this is the right word) aspecially on mobile devices to many click have to be made to get the effect

Legsmaniac, 2021-03-24:
Fair point.
Was just a suggestion that would at least work for now.
Not sure about cumbersome (yeah, good enough word), it takes mere seconds to pop in a couple of letters.
Anyhow, it'll be up to Aircookie to consider your suggestion.
It may well already be in the pipeline as I'm sure I remember seeing something mentioned a while back.
Pretty sure @Aircoookie will chip in and let you know when he has time.  😉

fishbone-git, 2021-03-24:
essentially similar request as #1262. also anoher suggestion/workaround mentioned there.

itCarl, 2021-03-24:
basicly yes

closed 2021-03-24 by itCarl