wled/WLED #1823: Missing / incorrectly set CSS style in index.css

SirColin, 2021-03-24:
`position: sticky;` move from `.lstI.sticky, .lstI.selected` to `.lstI`

previously
`.lstI {
overflow: hidden;
}`
`.lstI.sticky, .lstI.selected {
	position: sticky;
	z-index: 1;
}`

after
`.lstI {
  overflow: hidden;
  position: sticky;
}`
`.lstI.sticky, .lstI.selected {
	z-index: 1;
}`

Aircoookie, 2021-03-24:
Thank you for the suggestion :)
Why exactly is it invalid to set `position: sticky;` only on the `.lstI.sticky, .lstI.selected` class combinations?
The result seems equivalent, but I believe the current position makes more sense because only `sticky` and `selected` buttons should be sticking.

SirColin, 2021-03-25:
Hey :)

In Safari you will see the difference, the corners of the bar in the color preview are currently not rounded off, only with the sticky and selected classes it works.

the simplest solution is to move the `position: sticky;` to `.lstI.`

it can also be that I'm wrong and there is an even better workaround to solve the problem. :)

Legsmaniac, 2021-03-25:
> it can also be that I'm wrong and there is an even better workaround to solve the problem

Yeah, use a proper browser.  🤣

SirColin, 2021-03-25:
> Yeah, use a proper browser. 🤣

maybe :)

blazoncek, 2021-03-26:
`sticky` and `overflow` properties are not the best firends.
If parent object has `overflow` property set, `sticky` may have no effect in child elements.
https://stackoverflow.com/questions/43707076/how-does-the-position-sticky-property-work

Class `.lstI` has `overflow: hidden` set in line 916.

Aircoookie, 2021-03-26:
Thank you :) I just tested mobile Safari and can confirm that the overflow hide only works on the sticky elements. As, albeit counterintuitively, sticky still works correctly even when applied to the entire `lstI` class, I will implement your suggestion!

closed 2021-08-03 by stale[bot]