wled/WLED #1826: WledAp dont show up

insproad, 2021-03-26:
**Describe the bug**
A clear and concise description of what the bug is. Please quickly search existing issues first!

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

insproad, 2021-03-26:
I am using esp01 with  this https://drive.google.com/file/d/1_AzpjjOzXPZyEXoQzxX8QkZF-hsnWDMB/view. But when I try with esp12e it doesnt show up  wled-ap. How can I write my wifi id  and password to  this bin file.

WoodyLetsCode, 2021-03-29:
In your first post you say you use ESP32, in your second you say you use Esp8266.
Please clarify.
And maybe you can attach your _platformio.ini_ so we can see what settings you made.

WoodyLetsCode, 2021-03-30:
How did you flash it?

Aircoookie, 2021-03-30:
Hi! Please try `WLED_0.12.0_b5_ESP8266.bin` from this beta release https://github.com/Aircoookie/WLED/releases/tag/v0.12.0-b5 for an hopefully working WS2801 version on ESP8266 (note that you can set WS2801 type in LED settings, there are no longer LED type specific binaries in 0.12).

insproad, 2021-04-01:
> Hi! Please try `WLED_0.12.0_b5_ESP8266.bin` from this beta release https://github.com/Aircoookie/WLED/releases/tag/v0.12.0-b5 for an hopefully working WS2801 version on ESP8266 (note that you can set WS2801 type in LED settings, there are no longer LED type specific binaries in 0.12).

I tried  but  still not working. Wled-Ap doesnt show up.

insproad, 2021-04-01:
> How did you flash it?

I flashed with esphomeflasher.

xupernova, 2021-04-08:
Hi

I'm having issue with the wled AP not showing up after i switched off my ESP8266 for 1 day. 
WLED-AP will only show up each time after i reflash it. Not sure why the wled AP not showing up the next day or after switching it off for a long period.

Board: LoLin NodeMcu V3 ESP8266
Flash progam: ESPHome-Flasher-1.3.0-Windows-x64
WLED version: WLED_0.12.0_ESP8266.bin

xupernova, 2021-04-08:
> Hi
>
> I'm having issue with the wled AP not showing up after i switched off my ESP8266 for 1 day.
> WLED-AP will only show up each time after i reflash it. Not sure why the wled AP not showing up the [...]

I've enable the disable wifi sleep in the WLED settings before i switched it off

Aircoookie, 2021-04-08:
@xupernova hi, did you configure a connection to your home WiFi? In that case, the WLED-AP will no longer show up if the connection to your WiFi was successful. You can try setting the AP mode to "Always" if you would like the AP to be available at all times.

stewe93, 2021-05-21:
Hy @Aircoookie , Im. here to test anything related this issue because it affects me as well
NodeMcu ESP8266 (maybe V2 but im not sure about it) 
Tested binaries: 0.12.0, 0.12.0-b1

The AP barely shows up (tried from a macbook and from an android device) and if it is visible, doesnt accept the default password and after a few tries, the AP disappear. 
There was one time I successfully connected to the AP, but the initial config failed with an empty response 3-4 times, then it connected to my home network. 
When I reached it over my local network only the main page was accessible, everything else was empty response.

Tried to compile a binary for myself (nodemcuv2 profile), setting the network SSID and password in the firmware but it doesnt connect to my network then I think it fallbacks to AP mode in which im unable to connect to it.

Im using a very long ssid and password, I dont know if this could be a problem.

Im happy to compile and test anything you would like to see.

closed 2021-09-26 by stale[bot]