wled/WLED #1829: ws2801 doesn't work with esp32

bigu1975, 2021-03-27:
**Describe the bug**
esp32 board doesn't work with ws2801 strip

**To Reproduce**
Upload 0.12 beta bin file and connect ws2801 (with proper settings for LED e.g. data 2, clk 0). LED can't be controlled. Doesn't react on color/effect change, some of them are blank
used board: esp32 D1 Mini
tested pins: data 2, clk 0 and data 16 clk 4

**Expected behavior**
ws2801 strip connected to esp32 should work

**WLED version**
 - Board: ESP32 D1 mini
 - Version: 0.11.2 and 0.12 beta
 - Format: binary self compiled @ platformio (default_envs = esp32dev)

**Additional context**
@blazoncek advise It may be related to NeoPixelBus

At 0.10.2 it works quite better but even if leds can be controlled and are reacting on color/effect change from time to time some of leds flickering

on esp 8266 with 0.12 beta the same wiring works like a charm 

Thank you for your help!

sansillusion, 2021-03-27:
Are you using a level shifter ?
Sounds like data signal integrity issues.

bigu1975, 2021-03-27:
I have tested with shiftier and without. 
The thing is that the same wiring and soft works on 8266 device but doesn't on esp32
in addition changing only the led strip to apa102 on esp32 fix the problem as well.

huggy-d1, 2021-03-30:
I suggest using different pins for testing.  On 0.12.xß you know it's easy to move pins around.  Hopefully it's just dupont jumpers and not solder connections you are dealing with.

A reliable source suggests _only_ using these pins for digital outputs on ESP32:
2, 4, 12, 16-33

Perhaps your use of 0 & 2 got you headed in the head-down (sad) direction.  Pins 4 & 16 should have been good.

closed 2021-08-07 by stale[bot]