wled/WLED #1830: [enhancement] WLED deserves a website

tmeuze, 2021-03-27:
Hi @Aircoookie, 

I love this project. I was wondering if I can donate some money and time helping you set up a web frontend for the wiki? "wled.app", for instance, is $14USD per year. I have some experience with [MkDocs](https://www.mkdocs.org/) and [Material for MkDocs](https://squidfunk.github.io/mkdocs-material/) hosted on GitPages, and you could say I'm looking for a project to work on. 

Let me know if you have any interest. 

Thanks,

Aircoookie, 2021-05-17:
Hi!
I'm very glad that you like WLED so much that you would like to help me out in such a big way!
Sorry for my very late response.
A proper web frontend for the documentation would be absolutely fantastic!
My only condition would be that the content is easily updatable, either directly on-site, mirroring the existing wiki or hosted in a GitHub repo.
I also have a domain, `wled.me`, but the provider does not offer hosting, I can only redirect on my current plan, so `wled.app` sounds like it would be a great choice!

Let me know how I can help. Let's get this set up!

Thanks so much :)

tmeuze, 2021-05-17:
It's very simple and checks all your boxes! It can be hosted on GitHub pages and the content is markdown. It's just a nice frontend to the content you already have. 

I think the best approach to start it might be to copy your wiki contents to a new repo like "WLED-docs". Then people can contribute. All that would be needed is to [point your domain to the repo](https://docs.github.com/en/pages/getting-started-with-github-pages/configuring-a-publishing-source-for-your-github-pages-site), and it's published!

I PM'd you on your Discord if you have any questions. 

Thanks!

tmeuze, 2021-05-24:
Closing this as a repo is now available for the docs. Follow along at https://github.com/Aircoookie/WLED-Docs

closed 2021-05-24 by tmeuze