wled/WLED #1831: connecting analog rgb tape

mrgray777, 2021-03-27:
Hello everyone! tell me where can I find information on connecting analog rgb tapes? I cannot find such information, only incomprehensible pieces and general information. interested directly three-channel tape: red green blue

WoodyLetsCode, 2021-03-29:
Hello,
if you know the width of your led strip you can search for connectors.
Maybe you are looking for something like this: https://de.aliexpress.com/item/32242559186.html

mrgray777, 2021-03-30:
> Hello,
> if you know the width of your led strip you can search for connectors.
> Maybe you are looking for something like this: https://de.aliexpress.com/item/32242559186.html

you have a good sense of humor)) however, I'm interested in how to connect a three-color or monochrome tape to a WLED controller instead of an address tape. the wiki does not describe this method, although it is stipulated that it is possible. that's what I would like to know)

sansillusion, 2021-03-30:
You need MOSFET's to drive the current needed for analog LEDs, there
isnplenty of tutorial online for that. In wlet setup/led setup you can
chooses the type of LEDs and then the pins you wish to use. (0.12.x)

ericksvasc, 2021-04-15:
did you manage to do that?

closed 2021-08-28 by stale[bot]