wled/WLED #1833: WLED mit I2C möglich ?

tourer4778, 2021-03-28:
Hallo, ich bin noch nicht so tief in der ganzen Sache hier drin.
Ich nutze derzeit zwei Wemos D1 Mini in meiner Gartenhütte.
1 x I2C für Temperatur und Luftfeuchte erfassen
und 1 x für WLED.
Kann man das auch auf einen Wemos bekommen ?

WoodyLetsCode, 2021-03-29:
Ja das geht.
Schau dir mal das [WLED-wemos-shield](https://github.com/srg74/WLED-wemos-shield) von @srg74 an.
Da gibt es die Option 3 (I2C sensor). Dann hast du alles auf einem Board. 😄 

Und lies dir mal #658 von @Aircoookie durch.
Solche Fragen werden hier nicht mehr so gerne gesehen.

Aircoookie, 2021-03-30:
Genau, und für das Software-seitige lässt sich WLED prima durch Usermods erweitern. Für den BME280 Sensor etwa hat schon jemand einen Usermod geschrieben: https://github.com/Aircoookie/WLED/tree/master/usermods/BME280_v2

tourer4778, 2021-03-31:
Ich verstehe da leider nicht viel von.
Ich nutze derzeit fertige Firmware. Was ist ein Usermod ?

tourer4778, 2021-04-01:
Okay, danke

tourer4778, 2021-04-08:
Die wemos_shield_d1_mini_temperature_C.bin habe ich mir runter geladen.
Ich habe 2 x BME280 und  1 x HTU21 an dem einen Wemos angeschlossen.
Wenn ich das richtig gelesen habe, kann Shield nur BME280, oder?

srg74, 2021-04-08:
Diese Bin-Datei ist nur für den Dallas-Sensor bestimmt. Für Ihre Sensoren müssen Sie sich selbst kompilieren

tourer4778, 2021-04-08:
Okay, ich kenne mich nicht sehr gut aus in dem ganzen. 
Ich nutze derzeit Tasmota und da geht alles recht einfach, ohne etwas zu erweitern.
Von daher glaub ich, das es für mich hier nicht weiter geht.

closed 2021-08-22 by stale[bot]