wled/WLED #1835: configure wifi with a new router after use

sd0700, 2021-03-29:
Hi everyone, I'll explain my problem, I'm working on using the Wled for small shows, I made some LED tubes, but every time I go to a different place I would have to reconfigure the wifi with a new router, is there a way to reset the configuration and enter the new parameters without having to flash the board every time? I am using a wemos d1 mini pro, is there a way to create a reset button? thank you for your support

WoodyLetsCode, 2021-03-29:
Hello,
you can change wifi settings in Settings > WiFi setup.
Please don't use GitHub repository issue tracker to ask questions.
Read what @Aircoookie wrote in #658.

sd0700, 2021-03-29:
Good morning
my question was not this, I beg your pardon, my question is, if I have configured the wifi, and I change location with another wifi, how can I change the parameters if I have not deleted the old one before? is there a way to create a reset button, which deletes all the settings entered?
Thank you so much
Santo

WoodyLetsCode, 2021-03-29:
What exactly do you mean with _parameters_?

Legsmaniac, 2021-03-29:
The easiest way to do this would be to make sure you set up Configure Access Point and set AP Opens: Always.
Then when you are at a new location with a different router to connect to, just set your phone/laptop/computer to connect to the AP name you set in WLED AP Setup and use a browser to connect to 4.3.2.1
Then you can navigate to WiFi Settings and change the Network Name and Password you wish to connect to.

sd0700, 2021-03-29:
Thank you Sir 
I will try🙏🏼

Legsmaniac, 2021-03-29:
@WoodyLetsCode 

> Please don't use GitHub repository issue tracker to ask questions.

Then WHY when we open a New Issue and select Ask Question are we presented with.......

```
**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
```

I refer to the last bit....
**If you do not like to use these platforms, delete this template and ask away!**

Which implies it's OK to carry on and use this forum.

amcewen, 2021-03-29:
I've got a ["factory reset" usermod](https://github.com/mcqn/WLED/blob/master/usermods/factory_reset/usermod_factory_reset.h) in my repo which checks to see if pin 4 is held high for 5+ seconds at boot, and if so it wipes all the settings.

It uses a resistor and capacitor with the reset button on [my boards](https://github.com/mcqn/my-babys-got-led) so you can just hold the reset button down for 5 seconds to factory reset, but it should work with a second button if you wire it up to pull pin 4 high when it's pressed.

(I haven't submitted a PR for it, as it wasn't clear it'd be useful for others given it needs new circuitry, but I can if it's useful - it'd make pulling updates easier for me :grin: )

WoodyLetsCode, 2021-03-29:
@Legsmaniac I just read #658 and that is what @Aircoookie said.
I don't care. I only want to help to deal with these hundreds of issues.

sd0700, 2021-03-29:
Thank’s @amcewen and @Legsmaniac I will try both solutions🙏🏼

Legsmaniac, 2021-03-29:
> But maybe you are right. I won't write that anymore.

Thank you.  I would far prefer it to continue to stick to this forum and ask questions as I'm not a fan of Discord and hate having to join several forums all the time.  I know I'm not alone with this view.  There are plenty of members who can help answer questions, I'm sure, so there's little need for any questions to disturb Aircookie's time.

> I've got a ["factory reset" usermod](https://github.com/mcqn/WLED/blob/master/usermods/factory_reset/usermod_factory_reset.h) in my repo

Interesting alternative.  But isn't that effectively the same as the method I posted?  I mean, when you reset, you have to access 4.3.2.1 to set it up which is basically what I posted.  Your method will also reset all options - number of LEDs, Presets, everything whereas with my method, all settings remain intact.  😉

amcewen, 2021-03-29:
> Interesting alternative. But isn't that effectively the same as the method I posted? I mean, when you reset, you have to access 4.3.2.1 to set it up which is basically what I posted. Your method will also reset all options - number of LEDs, Presets, everything whereas with my method, all settings remain intact. :wink:

Indeed, mine does clear all the settings.  Figured it might work as a starting point to write a custom usermod.

I wasn't sure if yours required you to remember to change the settings each time before you visited a new network, but re-reading it looks more like a setting to have WLED create an AP whenever it can't find the one it's configured for?  That would make most sense for solving this.

Aircoookie, 2021-03-30:
Another option would be connecting a button to GPIO 0 and pressing that for 8 seconds, which will open the access point with default credentials.

About questions on Issue tracker: While I prefer the other channels, questions on here are certainly still welcome. Of course the amount of open issues is extreme currently, we need more stringent labeling to be able to sift through.

sd0700, 2021-03-30:
Thx @Aircoookie so I have to connect a button to GPIO 0 (pin D3 on Wemos) and ground? And pressing for 8 seconds?

closed 2021-08-07 by stale[bot]