wled/WLED #1837: 0.12.0.b4  Ethernet unable to get ip address over DHCP

bmcmahan123, 2021-03-29:
**Describe the bug**
When configured to use ethernet, the device never get's an up address

**To Reproduce**
Flash to the latest 0.12.b4 release.  Configure device over AP mode to use Ethernet (WT32-ETH0 in my case).  Plug device into ethernet and boot the device.  I can see the device on my network with an ip address of 255.255.255.255, however it never receives a valid ip address.

**Expected behavior**
When device is plugged into ethernet it should obtain an ip address and connect to the network.

**WLED version**
 - Board: WT32-ETH0
 - Version 0.12.b4
 - Format self-compiled

bmcmahan123, 2021-03-29:
Troubleshooting this a bit.  If I use the Espressif 32 2.0 instead of 3.3, everything works fine.  My example platformio_override.ini looks like the below:

[platformio]
default_envs = esp32_eth_debug

[env:esp32_eth_debug]
board = esp32-poe
platform = espressif32@2.0
upload_speed = 921600
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp32} ${common.debug_flags} -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE -D RLYPIN=-1 -D WLED_USE_ETHERNET -D BTNPIN=-1
lib_ignore =
  ESPAsyncTCP
  ESPAsyncUDP

Aircoookie, 2021-03-29:
Thank you very much for reporting!
We have run into https://github.com/espressif/arduino-esp32/issues/4778 here.
I will test `espressif32@3.2` and if it has no side effects, that will be used instead of `3.1.1`. Otherwise, a simple fix by replacing `INADDR_NONE` by `IPAddress(0,0,0,0)` in l. 156 of `wled.cpp` should work too.

commit 8755a9bc (2021-03-30): 0.12.0 Beta 5 (fixes #1837)
closed 2021-03-30 by Aircoookie