wled/WLED #1838: IR - Add on/off toggle code

mase76, 2021-03-29:
Hi!
I have a 44 key rgb remote, which is an usual one on Aliexpress. However this remote has a toggle code 0xff02fd for on/off. To use this remote, I have to build WLED with the custom on/off code. Maybe you could add a setting to configure a toggle code.

Aircoookie, 2021-03-30:
Hi! I do believe this is too specific to have a runtime setting for - but it might be a good enhancement to add a "Custom" IR type and allow reading of an `ircodes.json` file, which could enable WLED to adapt to all different sorts of IR remotes without requiring manual modification of `ir_codes.h` and a recompilation.

closed 2021-08-07 by stale[bot]

mase76, 2021-09-26:
I added a custom ir.json to the filesystem. However on/off does not work very well. I have to press the button multiple times to get a reaction. All other buttons work fine. When I add my code with #define IRCUSTOM_ONOFF into ic_codes.h, the reaction is fine. This is the on/off part of my json file:
`  "0xFF02FD": {
    "label": "Toggle on/off",
    "pos": "1x4",
    "cmd": "T=2",
    "rpt": true
  },`