wled/WLED #1839: Setup default color after lost power

pepsonEL, 2021-03-30:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

Hi
Is any place in GUI where i can defined default color which is start after power lost ? Now color start ORANGE....

Aircoookie, 2021-03-30:
If you have the preset you want to have at boot, remember its ID (left of the preset name) and enter it into the `Apply preset ... at boot` text field in LED settings

severendis, 2021-04-06:
> Wenn Sie die Voreinstellung haben, die Sie beim Booten haben möchten, merken Sie sich deren ID (links vom Voreinstellungsnamen) und geben Sie sie in das `Apply preset ... at boot`Textfeld in den LED-Einstellungen ein

Hallo,

genau das versuche ich einzustellen. Es funktioniert in der Version 0.12.0 bei mir auf einem D1 nicht. Ich habe mehrere PRESET erstellt. Die PRESET "1" soll nach einem Stromausfall gestartet werden. Dafür habe ich in den Einstellungen im Punkt `Apply preset ... at boot` die Ziffer 1 gewählt. Nach dem erneuten Anschluss der Stromversorgung bleibt der Stripe aus.

Wähle ich mit einem Klick die Einstellung PRESET 1, dann erhalte ich wieder die programmierte Sequenz.

Was bedeutet `(0) user defaults)`? Wo kann ich Default definieren?

severendis, 2021-04-06:
Das war die Lösung! Der Haken bei `0 use defaults` muss noch gesetzt werden.

Nun habe ich gleich noch eine Frage.
Ich habe 4 Presets konfiguriert. Alle statisch. Wechsle ich von einer Einstellung in die nächste (die auf 50% gedimmt ist) dann wird diese erst ganz hell und dimmt sich auf 50%. Kann ich das beeinflussen, das es eher von 0% auf 50% geht beim Wechsle des Preset?

Aircoookie, 2021-04-08:
Schön, dass es funktioniert hat!

Hmm, normalerweise sollte das Dimming immer von der aktuellen zur nächsten Helligkeit faden und niemals springen. Schick mir mal bitte die Datei `http://[Deine-WLED-IP]/presets.json` und nenne die beiden Presets bei denen das Problem auftritt.

severendis, 2021-04-08:
> 
> 
> Schön, dass es funktioniert hat!
> 
> Hmm, normalerweise sollte das Dimming immer von der aktuellen zur nächsten Helligkeit faden und niemals springen. Schick mir mal bitte die Datei `http://[Deine-WLED-IP]/presets.json` und nenne die beiden Presets bei denen das Problem auftritt.

Hier mal die Datei dazu.

[presets.zip](https://github.com/Aircoookie/WLED/files/6279051/presets.zip)

closed 2021-08-22 by stale[bot]