wled/WLED #1842: Instance list not working?

swampdogmash, 2021-03-30:
I have 12 b4 running on 2 devices. 've tried multiple configs of the wireless but no luck in getting one instance to show up on the other devices instance list. I thought maybe it was an IGMP snooping problem - but I've had that both on and off. I have also tried reducing the base rate to just 1 possibility - on the cisco APs broadcast traffic is sent over the base rate and not the main rate. None of these seem to have worked. I have run wireshark on a laptop on the same AP (SSID and physical access point) and can't see any packets being sent from the WLED instances when opening the instance list - I can however manage the devices fine.

They are running on a wemos d1 mini and a wemos d1 mini pro.

Aircoookie, 2021-03-30:
Hi!
Instance list discovery packets are sent via broadcast on IP 255.255.255.255 UDP port 65506. No packets are explicitely sent when opening the Instance list UI, all controllers just transmit a broadcast every 30 seconds.

There are multiple reasons why it might not work:
- Instance list and/or instance discovery are turned off in Sync settings
- The "2nd port" in the WLED broadcast section in Sync settings is not the same on both units (default 65506).
- The unit which is to be listed has the WLED-AP access point enabled (this is unwanted behavior and will be fixed)
- Network configuration not distributing broadcast messages. You can also test the regular WLED sync, if that doesn't work, it is likely not an issue with Instance list itself.
- A bug in WLED/Instance discovery

swampdogmash, 2021-03-30:
Thank you for the comprehensive reply. That gives me plenty to test - and find the problem :)

1. Instance list and/or instance discovery are turned off in Sync settings
            _checked both instances and they both have both boxes ticked_
2. The "2nd port" in the WLED broadcast section in Sync settings is not the same on both units (default 65506).
            _Both had primary AND secondary set to 65512 - have changed to pri 65512 and sec 65513. for good measure I have rebooted both units also_
3. The unit which is to be listed has the WLED-AP access point enabled (this is unwanted behavior and will be fixed)
            **Both units have "WLED-AP" in the AP config and have Hide AP Name unticked. I can't see them, but does the AP only appear when the device is NOT connected to wifi? Either way - have just cleared the AP name box and all working now :)**
4. Network configuration not distributing broadcast messages. You can also test the regular WLED sync, if that doesn't work, it is likely not an issue with Instance list itself.
            _Regular sync is working fine both before and after the change to the pri and sec ports above. On a laptop on the same wireless SSID and subnet I can see the broadcast packets for sync without issue._
5. A bug in WLED/Instance discovery
          See number 3! :)

One question - If I wish to sync devices in separate groups by changing the primary port (same port in each room for example), can i still keep the secondary port the same to allow all instances to be shown in the instance list without causing them all to sync together?

Aircoookie, 2021-03-30:
Awesome that you got it to work! I suspect the issue happens if the AP was active after boot before connection to your network was established. This is controlled with the "AP opens" dropdown. With the default "No connection after boot", it could be possible that the ESP did not connect in time and thus momentarily opened its AP.

> One question - If I wish to sync devices in separate groups by changing the primary port (same port in each room for example), can i still keep the secondary port the same to allow all instances to be shown in the instance list without causing them all to sync together?

Of course! That use case is the reason for the instance list using the secondary port.

swampdogmash, 2021-03-30:
Fantastic, thank you for a PROPER testing list and for keeping on top of so many things! Awesome stuff!

swampdogmash, 2021-03-30:
I think maybe it was both. I changed them in order listed with a reboot in between each.... and it didn't work after changing the secondary port. It only worked when i changed the AP name to empty and saved and then rebooted.

closed 2021-08-07 by stale[bot]