wled/WLED #1845: Wrapping Animations Bug

dgwood12, 2021-03-31:
**Describe the bug**
When testing a few animations, I encountered a bug today with the Police effect and linear wrapping. The strip lags when wrapping the animation and the animation wont show up on the first LED until about the time it has finished on the last LED and when it finally does, it catches up with where it was supposed to be (or so it appears). It is more prevalent when the effect intensity slider has a higher value but still has a slight glitch at lower intensities as well. 

**To Reproduce**
Select Police Effect, make sure linear palette blending is enabled and select a higher effect intensity

**Expected behavior**
The strip lags when wrapping the animation and the animation wont show up on the first LED until about the time it has finished on the last LED and when it finally does, it catches up with where it was supposed to be (or so it appears). It is more prevalent when the effect intensity slider has a higher value but still has a slight glitch at lower intensities as well.

**WLED version**
 - Board: Wemos D1
 - Version 0.11.1
 - Format: Binary

**Additional context**
Here's a video showing the issue: https://imgur.com/a/X7BDsel

Thank you for your help!

Legsmaniac, 2021-03-31:
I had a go with this.  I could only kind of reproduce it if I set colour #2 to either red or blue.
Any other colour or left as default (black) it runs as expected for me.
Then again I was testing on version 12.0.b5

dgwood12, 2021-03-31:
> I had a go with this. I could only kind of reproduce it if I set colour #2 to either red or blue.
> Any other colour or left as default (black) it runs as expected for me.
> Then again I was testing on version 12.0.b5

I just upgraded to 12.0.b5 and am having the same issue as before. I checked and have no color set for #2 (black). Did you try to run it at a slower effect speed? It looks somewhat normal at a fast speed but I suspect that's just because its going too fast to notice. Run it with an effect speed of 10 and effect intensity of 100. I just tested this and the problem is occurring with these values.

Legsmaniac, 2021-03-31:
Ah yes, gotcha.  I see it now.  Deffo gotta be a bug.

closed 2021-08-07 by stale[bot]