wled/WLED #1846: Unable to some LED Preferences including (but not limited to) output type and output Pin using iOS native app.

dywicked, 2021-04-01:
**Describe the bug**
Inability to change some LED Preferences including (but not limited to) output type and output pin using the iOS native app. This has been reproduced by @crasher1964 on discord - please see: https://discord.com/channels/473448917040758787/779395228624617512/826963786128162816

**To Reproduce**
Using iOS Native app, navigate to Config -> LED Preferences and change the LED output type or Pin. Then attempt to save the settings - upon tapping save nothing happens. @crasher1964 suspects the issue may be broader and may be limited to numeric or enumerated types. After making you

**Expected behavior**
Settings should save when tapping save - in v0.11.1 this works without issues.

**WLED version**
 - Board: Wemos Mini D1 ESP32
 - Version: v0.12.0-b5
 - Format: Precompiled WLED_0.12.0_b5_ESP32.bin from this repo.

**Additional context**
No issues in any browsers I've tested including Chrome on Mac, MobileSafari and Chrome on iOS.

Aircoookie, 2021-04-06:
Should be fixed in final 0.12.0 release :) Please reopen should the issue still occur.

closed 2021-04-06 by Aircoookie

dywicked, 2021-04-07:
I tested as soon as v0.12.0 was released, all looks good.