wled/WLED #1848: Alexa discover problem on NodeMCU

Cosmicbase, 2021-04-01:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

Hello everybody,

 I use WLED on 3 different NodeMCU.
 Always the latest release.  After I have deleted the device (s) from Alexa, they are no longer found.  Tried everything from restart to reinstallation.  What's wrong?

 Please help

Cosmicbase, 2021-04-02:
I have now reinstalled everything, but on a Wemos D1 Mini. And here we go, the Alexa emulation works. I think that the Nodemcu is only suitable for this to a limited extent.

VoidElle, 2021-04-02:
I have the same problem with a NodeMcuV3,
I have the alexa option active but alexa doesn't detect it!

I also tried to fix:
Restarting alexa
Restarting the nodemcu
Checking if the network is 2.4Ghz in both the devices

etlon, 2021-04-02:
I have the same problem. NodeMCU ESP8266 with WLED but no working alexa emulation.

etlon, 2021-04-02:
@Cosmicbase are you using the d1 mini esp8266 or the d1 mini esp32?

Cosmicbase, 2021-04-03:
I use Wemos D1 Mini with Esp8266 chipset.

VoidElle, 2021-04-03:
> I have the same problem with a NodeMcuV3 ESP8266,
> I have the alexa option active but alexa doesn't detect it!
> 
> I also tried to fix:
> Restarting alexa
> Restarting the nodemcu
> Checking if the network is 2.4Ghz in both the devices

The stable version 0.12 fixed this problem for me.

Cosmicbase, 2021-04-03:
The stable version 0.12 fixed this problem for me too.

closed 2021-04-03 by Cosmicbase