wled/WLED #1849: WS2812-C

SexyRoyalBlue, 2021-04-02:
Will there be any compatibility with the WS2812-C?  It would be very useful to me

Aircoookie, 2021-04-02:
Hi, I would expect WS2812-C to be compatible with the WS2812B protocol. You can certainly give it a try :)

SexyRoyalBlue, 2021-04-02:
I actually have five sets of them on a pcb here and I tried them. A friend of mine told me that the time to "speak" to them is diffrent from B to C and it proves this point with weird rainbow flicker when it just should be plain solid orange

Aircoookie, 2021-04-03:
I've just confirmed in the datasheet that WS2812C indeed requires different timings unfortunately. Seems like 2812B should be compatible with 2812C protocol, but not the other wau around due to the huge reset delay required by 2812C.

Code | WS2812B | WS2812C
| --- | --- | --- |
T0H | 250ns~550ns | 220ns~380ns
T1H | 650ns~950ns | 580ns~1µs
T0L | 700ns~1µs | 580ns~1µs
T1L | 300ns~600ns | 220ns~420ns
reset | >50µs | **>280µs**

You could raise this to https://github.com/Makuna/NeoPixelBus , once a new type is supported there, it can easily be added to WLED, too.

SexyRoyalBlue, 2021-04-04:
So Makuna suggested using or trying out NeoWs2812xMethod which has the longer reset time already

Aircoookie, 2021-04-07:
Thanks! What LED pin are you trying to use and are you running on ESP8266 or ESP32?

Most pins use a variant of `NeoWs2813Method`, which is an alias for `NeoWs2812xMethod`. If you use ESP8266 gpio 2 or 1 or ESP32, that correct method is used already since it yields the most compatibility. Troubleshooting this issue is likely easiest if you installed Arduino IDE and flashed one of the NeoPixelBus example sketches to see if the issue also occurs with that.

SexyRoyalBlue, 2021-04-07:
I will try to flash the sketches, thanks 
And yes, I'm running an 8266

closed 2021-08-22 by stale[bot]

DarthWeber, 2023-10-30:
Can somebody confirm if those WS2812-C are supported, especially in any combination with ws2812-B on different pins of an esp32?

blazoncek, 2023-10-30:
Check with NeoPixelBus.
If the use different constructors (i.r. driver) than WS2812B then they need to be implemented in WLED.