wled/WLED #1852: V0.12 APA102 ESP8266

r100gs, 2021-04-03:
Cannot Set CLK to 0 ist Always stayes at 2 

Set Data to 2 and CLK to 0

CLK should bei 0

**WLED version**
 - Board: Wemos D1
 - Version 0.12.0
 - bin downloaded

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

Aircoookie, 2021-04-03:
Thank you for reporting!
I just tried to reproduce and can set CLK pin to 0 successfully for APA102 in 0.12.0. If you get a `Pin Conflict!` message, clear the button pin field to `-1` (it uses pin 0 by default). If it let's you save the LED settings, but the pin is still not set correctly, please send me your `[WLED-IP]/cfg.json` file so I can figure out if the bus is configured correctly :)

r100gs, 2021-04-03:
|  
-- | --
rev |  
0 | 1
1 | 0
vid | 2012180
id |  
mdns | "wled-6910fc"
name | "WLED-WOZI-TV-APA102"
inv | "Light"
nw |  
ins |  
0 |  
ssid | "SEMSA"
pskl | 24
ip |  
0 | 0
1 | 0
2 | 0
3 | 0
gw |  
0 | 0
1 | 0
2 | 0
3 | 0
sn |  
0 | 255
1 | 255
2 | 255
3 | 0
ap |  
ssid | "WLED-AP"
pskl | 8
chan | 1
hide | 0
behav | 0
ip |  
0 | 4
1 | 3
2 | 2
3 | 1
wifi |  
sleep | true
phy | 1
hw |  
led |  
total | 196
maxpwr | 11000
ledma | 55
rev | false
rgbwm | 3
ins |  
0 |  
en | true
start | 0
len | 196
pin |  
0 | 2
1 | 2
order | 0
rev | false
skip | 0
type | 51
btn |  
ins | []
ir |  
pin | 4
type | 0
relay |  
pin | 12
rev | false
status |  
pin | -1
light |  
scale-bri | 100
pal-mode | 0
gc |  
bri | 1
col | 2.8
tr |  
mode | true
dur | 7
pal | 0
nl |  
mode | 1
dur | 60
tbri | 0
macro | 0
def |  
ps | 0
on | true
bri | 20
if |  
sync |  
port0 | 21324
port1 | 65506
recv |  
bri | true
col | true
fx | true
send |  
dir | false
btn | false
va | false
hue | true
macro | false
twice | false
live |  
en | true
port | 5568
mc | false
dmx |  
uni | 1
seqskip | false
addr | 1
mode | 4
timeout | 25
maxbri | false
no-gc | true
offset | 0
va |  
alexa | false
macros |  
0 | 0
1 | 0
blynk |  
token | ""
mqtt |  
en | false
broker | ""
port | 1883
user | ""
pskl | 0
cid | "WLED-6910fc"
topics |  
device | "wled/6910fc"
group | "wled/all"
hue |  
en | false
id | 1
iv | 25
recv |  
on | true
bri | true
col | true
ip |  
0 | 192
1 | 168
2 | 88
3 | 0
ntp |  
en | true
host | "192.168.88.1"
tz | 2
offset | 0
ampm | false
ol |  
clock | 0
cntdwn | false
min | 0
max | 29
o12pix | 0
o5m | false
osec | false
timers |  
cntdwn |  
goal |  
0 | 20
1 | 1
2 | 1
3 | 0
4 | 0
5 | 0
macro | 0
ins | []
ota |  
lock | false
lock-wifi | false
pskl | 7
aota | true
um | {}

ggitthub, 2021-04-03:
Can confirm for ESP8266 with WS2801 setting (previously flashed with WLED_0.11.1_ESP8266_WS2801.bin). Data and Clock are both set to 2 by default. I set CLK pin to 0 (and move button pin to 4 or something, to solve pin conflict message), saved, CLK reverts to 2.

Cosmicbase, 2021-04-03:
I have the same problem. I changed the button pin to -1 an clk 0, data 2. After I saved the config, the clk returned to clk2.

Cosmicbase, 2021-04-03:
Gute Neuigkeiten. Habe auf APA102 gestellt Data 2, Clock 0 und Button Pin -1. Neustart und nochmals überprüfen. Bei Bedarf nochmals die Pins setzen, speichern und Neustart.
Clk bleibt auf clk0. 😋

Aircoookie, 2021-04-03:
I could reproduce the issue.
The problem is that WLED 0.11 incorrectly saved the data pin twice for 2-pin LEDs, instead of data and clock as it should have been.
WLED 0.12 realizes that [2,2] is an illegal pin config and thus doesn't init the bus. This leads to the LED count in the bus config in LED settings being set to 0. Even after correcting the pin issue, the bus is not saved, since it thinks it has 0 LEDs.
This is only an issue with instances that ran 2-wire LEDs in 0.11, it does not occur on fresh 0.12 installs.

**As a workaround, set the clock pin to 0 and enter the LED count into its field again (the per-bus LED count, not the total in the very top). Now the bus should save correctly and work.**

ggitthub, 2021-04-03:
Yep, worked for me. Data = 4 or 2 (tried both), Clock = 0, Start = 0 and Count = total LEDs on the (single) WS2801 strip, same value as on the top.

r100gs, 2021-04-03:
For me ist works too