wled/WLED #1853: Colour Control/Effects control iOS Browsers, maybe android? not checked.

Doyle4, 2021-04-03:
**Describe the bug**
Cant Select Effects or Colours on iOS, Google chrome/Safari, can see the button turn a lighter shade but no white spoit appears or effect/colour activated.

**To Reproduce**
Clear cache for new UI, try to select a colour or effect

**Expected behavior**
Change Colour or effect

**WLED version**
 - Board: D1 Mini
 - Version 0.12.0 New release
 - Format Pre made from Releases/Custom

**Additional context**
Possible to add the option to activate old UI in settings? 

Thank you for your help!

Aircoookie, 2021-04-03:
Hi, sorry about that and thank you for reporting!
There should be a little dot in the bottom left corner of the UI, is it green, yellow or red? It should normally be green and turn yellow for a split second when you try to select another effect or color.

I have tried to reproduce the issue, but both the color wheel and effect selection works fine for me in WLED 0.12.0 on iPad 6th gen. + both the app and Chrome on iOS 14.1. What device are you running on?

Doyle4, 2021-04-03:
The Green dot stays green.
Im currently out at the moment but when home later i’ll record my screen and show you.

Im on iPhone iOS 12.4

brianmiller1956, 2021-04-03:
Hi, 
Just to confirm I have just tried this on an iphone6 on iOS 12.4, neither safari or chrome browsers are working on wled12, nor the iOS native WLED app, all 3 give no response to effect or colour change,

Works fine on my iPad pro and iPhone8 on iOS 14.4
Safari, chrome and iPad WLED app all ok.

Doyle4, 2021-04-03:
Thanks for confirming this @brianmiller1956 , I'm wondering if its to do with the iOS or Browser versions.

brianmiller1956, 2021-04-03:
Hi, Because it affects the native iOS app WLED as well as both browsers it looks like an iOS 12 related problem. all 3 have the same problem with WLED 12, but all work with WLED 11 without problem

Doyle4, 2021-04-03:
Fingers crossed it can be resolved for iOS 12, I enjoyed the UI experience on the android device, but its not mine sadly.

brianmiller1956, 2021-04-03:
It’s actually wled12 that has the problem, wled 11 does work with ios12.

Doyle4, 2021-04-03:
True, Didn't see it that way.

msi2980, 2021-04-04:
I am having what seems to be the same issue on iOS 12.5.1 iPad Air and my iPhone 7. I can change strips to the 1st color in palette but nothing else. Can cycle thru the effects and color choices with no response. Have to very gently touch the selection of preset color or effect where the little dot appears. Using a dig quad and have tried on a couple different new boards

msi2980, 2021-04-04:
I also had no issue with wled 11

Doyle4, 2021-04-04:
No issue with 0.11.1 on iOS 12.4

0.12 Stable release, can change colour using the default dots, enter different settings such as effects, config etc, just not Colour options or effects.

Doyle4, 2021-04-04:
https://user-images.githubusercontent.com/65953502/113514084-8c6aa680-9564-11eb-8be2-60e6b5a9c786.MOV

msi2980, 2021-04-04:
I’m able to select the different colors by gently tapping on iPad where the dot goes ever so lightly but the indicator light bottom left stays green and no response

Aircoookie, 2021-04-06:
Thank you for the confirmations!
Unfortunately I do not have an iOS 12 device to my disposal to pinpoint this issue, albeit I have a hunch of what could be causing it, namely the sticky property of all the effect + palette buttons.

Could you try if the issue is also present in the 0.12.0 Beta 2 build attached below?
https://drive.google.com/drive/folders/1O7yFqSZKCgUy_2-2_4mNrr21XYwWepMe?usp=sharing

brianmiller1956, 2021-04-06:
Hi, I have just tested the ESP8266 version and it has the exact same problems, 
Tested using Safari, Chrome and native iOS WLED app,
The problem is identical no actual change to palette or effect selection,  green dot stays green, the colour wheel does set colours 1,2&3

msi2980, 2021-04-06:
Damn I’m sorry but  I was actually due to upgrade iPad , which I did and I
already updated my iPhone so I don’t have access to iOS 12 anymore.

Doyle4, 2021-04-07:
> Thank you for the confirmations!
> Unfortunately I do not have an iOS 12 device to my disposal to pinpoint this issue, albeit I have a hunch of what could be causing it, namely the sticky property of [...]

Tested, can confirm no change. iOS 12.4 iPhone X

JensMKofoed, 2021-04-25:
Hi

I can confirm there are a problem with newer version of IOS 12. At some point i have updated my old  IPhone 5S and it will no longer change color palates or effect to wled. It doesn't matter if it is via a browser (Safari/Chrome/Firefox) or the native wled app.
today I updated the IPhone to IOS 12.5.2 and it stilldoesn't work with wled. I have the same problem with an IPad also running IOS 12.4.8
The problem is not only with wled version 0.12 but also with 0.11
regards
Jens

Doyle4, 2021-04-25:
@JensMKofoed 
After you flash back to 0.11 you need to reset your caches on iOS devices, it will then return back to old UI :)

JensMKofoed, 2021-04-25:
I didn't flashed back to 0.11. I have 2 devices with different firmware. I just checked that the issue are with both devices/firmwares.

blazoncek, 2023-09-19:
@Aircoookie is this still relevant or should we close?

Doyle4, 2023-09-20:
Was fixed a while back.

closed 2023-09-20 by Doyle4