wled/WLED #1854: drip effect reboots system when selecting reverse option in segments

dr44dst44l, 2021-04-03:
when running the drip effect, choosing "reverse direction" in the segment section causes wled to reboot

**To Reproduce**
select the drip effect
go to segment section and choose reverse direction
watch wled reboot

**Expected behavior**
reverse direction of the effect

**WLED version**
 - Board: Esp32 from Quindor
 - Version 0.12.0 dev2104020
 - Format Binary WLED_0.12.0_ESP32.bin

**Additional context**
FVDB could reproduce the problem
tested on Dig_Quad with four ledstrips on four channels divided in four segment`s and
on normale esp32 Wroom single ledstrip single segment

filipvdb, 2021-04-03:
Tested with a QuinLED-Dig-Quad board with ESP32 from Quindor and generic ESP32 modele D1mini format. When changing the effect to drip and selecting reverse on any segment, the ESP module reboots and comes back with the default WLED orange color.

Up till now only found this issue with the drip effect.

Aircoookie, 2021-04-03:
Can confirm. Issue is ESP32 specific. Will be fixed soon, thank you :)

Aircoookie, 2021-04-03:
Hi, this problem should now be fixed :)
Please try this binary and report back whether it works!
https://drive.google.com/file/d/1ViAxbVY-n5zRqSYBroawCg-GQR6AgDKD/view?usp=sharing

filipvdb, 2021-04-03:
Confirmed for me on 2 boards running an ESP32

closed 2021-04-03 by Aircoookie

dr44dst44l, 2021-04-03:
confirmed for me too

blazoncek, 2021-04-06:
Interesting enough without any change to drip effect my fork does not cause ESP32 to reboot.