wled/WLED #1856: Update from 0.11.1 to 0.12.0 fails

ggitthub, 2021-04-03:
Hello

I updated two devices successfully to 0.12.0, but a third controller fails repeatedly ("Failed updating firmware! Please check your file and retry!“). The failing hardware is the "WLED Pixel Controller" from codm: https://shop.codm.de/automation/pixel/30/wlan-pixel-controller-wled (it worked on two different hardware types).

I tried with reboot and also removed the power physically, used two different browsers, but always the same result. Instead of WLED_0.12.0_ESP8266.bin I tried also the already installed (v.0.11.1) and the 0.12.0_b5 firmware, same outcome.

I did not try the factory reset because I don't want to lose 30+ presets...

Could it theoretically be the settings (LED count/type, presets, etc.), or is that rather unlikely?

unterleib, 2021-04-03:
Hi, have the same issue, updated the first wemos d1 mini with OTA, after the update successfully message the browser refreshes with "can not connect....."  blank page.
No change after taking the wemos d1 mini of the electricity. 
Tried even a second on, same thing here.

Aircoookie, 2021-04-03:
Hi!
I believe the firmware that came preloaded on the codm Pixel Controller utilizes a different partition layout (`2m512k`) than standard WLED binaries (`4m1m`). OTA updates to a binary compiled with a different partition layout are not possible.

I've compiled a 0.12 binary with the 2m512k partition layout for you. You can try if the update works with that! Just to be save, open `[WLED-IP]/presets.json` beforehand and copy its contents to a text file. That way, all your presets are backed up should anything go wrong.
Binary: https://drive.google.com/file/d/173-WLmZOrOONrOtBK4wVG6fflKInyR08/view?usp=sharing

ggitthub, 2021-04-03:
Aircoookie is the hero! The update worked! Donation sent.

Why is codm using that different layout? And that will prevent future updates as well, right? It will always require a special firmware?

codmpm, 2021-04-03:
> Why is codm using that different layout? And that will prevent future updates as well, right? It will always require a special firmware?

That's just historically "grown". Of course you could reflash (OTA, USB-Serial, etc.) the standard binary to have the correct filesystem layout - sadly not via the webinterface.
Feel free to contact me via shop@codm.de so I can reflash the board for you if needed.

sansillusion, 2021-04-03:
> Feel free to contact me via ***@***.*** so I can reflash the board for

You can download a backup of your presets if you go to
http://YOURWLEDIP/edit
Since when you reflash using standard partition format trough USB or OTA
you will loose your presets/settings.

closed 2021-04-07 by Aircoookie