wled/WLED #1857: WLED V0.12.0 bug report

djwikkid, 2021-04-03:
Soon as I saw it was released I flashed it to my nearest D1 Mini and this is what I found. I did read the feature update list so I knew what to expect and look for. Here's what I didnt find.

-Effect preview is not present. I saw how it should be shown only after watching Quin's update he put out today. 

-New effects won't work twice. I was eager to see the new effects TV and the other T effect name that escapes my mind writing this, but first time I clicked on each of the 4 new ones, the strips wont go back to it a second time without resetting the board. 

I am using a D1 Mini Dev Board and unsure of the which manufacturer mine came from so I took pics below. The Strip is a RGBIC 144/m 3.3ft strip from Alitove purchased through Amazon. Running Win10 and interfaced via Chrome browser at time of bug smacks. This is my first ever bug report and just trying to be thorough and give the best info about what I found. I also read the Nov 4th2020 pinned post about the growth of the community and it being too much for just one person to do all the things Aircookie wrote about.  I totally understand 100 and feel blessed he...or you depending on who is reading this.... was able to respond to an email I sent a couple of months ago when I was getting my feet wet so to speak. Please let me know if there's any other info needed I overlooked. TYVM, Dan

[image]

[image]

djwikkid, 2021-04-04:
Later the same day I did an Manual OTA to a installed board. None of the issues I listed above happened and the board used is the same type of D1 Mini just a different board than was used earlier. I am just now seeing how horrible those pics are above and remind you this is my first official bug report ever other than just being a highly critical human to something that was not mine. I hope there is enuff visual cues so someone can figure who made the thing.  Sorry not sorry. But thanks always.

Dan

pfeerick, 2021-04-06:
Hey, thanks! Any, and all reports are helpful, if it's a problem that developers can reproduce or can identify the cause of. 

When you say `Manual OTA` ... do you actually mean you flashed WLED onto the board via the USB? As that isn't OTA (Over The Air) ;) Also, when you take phones in future, either try doing it with the board flat on the table, so that everything in the shot is at the same level or tap on the screen to make it focus on the object you're interested in (it's that badly out of focus because your camera focused on the can in the background instead of what you were interested in). 

I can't say I had this issue while we were testing(as most of my WLED stuff is ESP8266, rather ESP32 like a lot of the other beta testers). I also can't repeat it on my setup at present... most of my boards are Wemos D1 Mini clone like yours, and I just tried with a 20 WS2811 LED setup running 0.12, and was able to switch between the four new effects (Aurora, Dynamic Smooth, Tetrix and TV Simulator) and others repeatable without need to reboot the ESP8266 board.

Aircoookie, 2021-04-06:
Hi and thank you for the report!
If the palette preview does not show up and you can't switch effects properly, that usually indicates some form of connectivity problem. Are you running the board in `WLED-AP` mode or connected to your network? If the latter, what is the signal strength reported in the Info page? There is a little colored dot in the bottom left of the UI, it should be green and briefly turn yellow if you click something.

Just had another thought: If you ran a 0.12 beta or any dev version after 0.11.1 on this board, there is a possibility that Chrome cached an old version of the UI (that bug is fixed in 0.12, but not in retrospective). You can `Ctrl + F5` to force a reload or try it with a different browser.

djwikkid, 2021-04-07:
Thank you for the FFW the Tips I would figured out if you as a couple days.
The pictures were took with a webcam I just got and yes I read the mt dew
can she couldn't make out. Can you believe there is that Much sugar in a
12oz can of soda?
I can't explain or repeat the report I made for this bug and when I spoke
with Aircookie, he thinks it was Kind of connectivity issue which means I
messed up in a small way because the soldering.g I did was very steady and
has minimal flux used. So I made a small mistake...my Gartner would get
animated right now and day shocker and sound like half of the joker.

I do thank you for writing back. If you know any tips for making wled react
to sound, plz fill me in before you hit your next stop. 1st mistake was
buying a bag of just microphones,nothing else. Reading how microphones
signal has to be amplifiers so the 0 crossing of the sine wav up so that
the bottom half of the wad can be read and made into a control signal, I
went back to the discord and think the max9814 setup is my best option.
Thanks for not going to hard on me. Lol Hard on...so sorry. I'm a46  year
old juvenile in brief spurts...that's what she did... Go forward in
blessings. And will pray to him that I do much better next time to prove my
value or some value to th3 WLED project.

Dan H

djwikkid, 2021-08-07:
Hey Christian, I didn't know this was open either. Go ahead and close it
since the upgrade with be released soon. I do have a new small issue with
my app on Android, but I'll go through the protocol on discord or Github.
Hope things are cranking for you, but found that extra bit of time you been
need for yourself you so deserve.

Danny

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
08/07/21,
02:23:50 PM

closed 2021-10-12 by djwikkid